dresden-elektronik/deconz-rest-plugin #2137: [Request device support] Aqara Opple 3fach Switch (WXCJKG13LM)

DocGame69, 2019-11-27:
I wrote about my problem in # 2120. Unfortunately in German (my Endlisch is misserabel).
Here is the Google Translate translation :-):
I get the Aqara Opple 3x Switch (WXCJKG13LM) Aqara Opple Switch not taught in Phoscon and displayed. However, the Osram power socket I used to learn switches on and off with each press of the button. The button is not displayed in Phoscon or in the Deconz or in the Iobroker-Deconz adapter. He must have connection, because the socket switches.
Exactly the same behavior I have with the new Ikea dimmer switch.
Are the devices still being maintained? I need these for some projects.
The 2.05.70 (Windows 10) is installed. Unfortunately, version 2.05.71, which is written here, can not be downloaded as an .exe.

Greeting DocGame

addellll, 2019-11-27:
The Aqara Opple switch already have an thread:
#https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061

closed 2020-04-02 by stale[bot]

mapsrof, 2020-04-05:
Same problem here. WXCJKG13LM is displayed as device in Phoscon, but you can't select any buttons.

Tedious01, 2020-04-08:
Push - same here...

helipus, 2020-06-20:
Push same here. WXCJKG13LM is connected but pushed buttons are not shown.

SwoopX, 2020-06-20:
This is not an issue of the REST API, but of Phoscon. please raise a corresponding issue here https://github.com/dresden-elektronik/phoscon-app-beta