dresden-elektronik/deconz-rest-plugin #2141: Hue Tap not working / not pairable again

michis0806, 2019-11-28:
Hi,

my Hue Tap hasn't worked since yesterday. None of the keys performs any function. When I decoupled the tap and wanted to reconnect it, I couldn't find that the "normal" dimmer is available under Switch > Philips, but the tap is grayed out.
[image]

Does anyone have an idea how to fix this?

Thanks
Michael

johanneswilm, 2019-12-06:
Same here. I was actually able to pair it by right clicking the philips tap image, clicking on "inspect" then removing the "disabled" from the class of that element. After that I could click it and pair it like any other switch. However, after setting it up, it seems like the buttons don't actually have any effect on anything.

Any clues, @manup ?

johanneswilm, 2019-12-06:
Update: the switch is actually working now, so it seems to just be a minor issue with the web app adding that "disabled" class. I'm guessing it's something left over from testing, maybe?

michis0806, 2019-12-09:
I used the "other" way to connect the Tap again and that worked...

johanneswilm, 2019-12-09:
What is the "other" way?

michis0806, 2019-12-09:
Add new switch => "Other" ;)

closed 2020-04-14 by stale[bot]