dresden-elektronik/deconz-rest-plugin #2142: Cannot disable WiFi

tomasrudh, 2019-11-28:
I have deconz 2.05.66 running on a Raspberry Pi with Raspbian 10 Buster. I have WiFi disabled in /boot/config.txt. I have stopped deconz using systemctl stop deconz-wifi and disabled using systemctl disable deconz-wifi. Nevertheless, after a couple of days is deconz-wifi running again, filling syslog with error messages. Is there a way to permanently turn off deconz-wifi?

Smanar, 2019-11-29:
You have perhaps your solution here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1738

tomasrudh, 2019-12-10:
An update of deconz solved the problem with log spam. I have also disabled the daemon deconz-wifi because I do not use it, my resources are a bit sparse. I am curious as to why the daemon is enabled again automatically, and the daemon started.

tomasrudh, 2020-01-18:
Is there any progress to this case? Anything I can do to turn off deCONZ WiFi?

closed 2020-05-24 by stale[bot]