dresden-elektronik/deconz-rest-plugin #2143: Contact Sensor Rules - Door OPEN or CLOSED

gomensa, 2019-11-28:
Hi 
I am a newbie so not sure if this is the right place to address this. 

Is there contact sensor rule support in Phoscon?

I have added a Xiaomi door / window sensor to Phoscon. It shows up in the sensors list and displays when the door is open or closed. However, I cant add a sesnsor rule to a group for the door sensor to control lights.

Smanar, 2019-11-30:
Not in phoscon yet, but you can take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1674

gomensa, 2019-12-05:
Awesome. Got it to turn on lights when door is open.

Where is the best place to get forum help on this topics? I am having trouble with a rule to transtion the lights to off when the door is closed. I have a motion sensor in the hall which gets activated fine, just it does not pick up motion by the door. Its too far away. So once the door is closed i want to turn off the hall lights but not immediately.

I tried the below but it turns off immediatly when the door is closed

{
	"actions": [
	{
        "address": "/groups/7/action",
        "body": {
            "on": false
            "transitiontime" : 1200
        },
        "method": "PUT"
	}
	],
	"conditions": [
	{
		"address": "/sensors/23/state/open",
		"operator": "eq",
		"value": "false"
	},
	{
        "address": "/sensors/23/state/open",
        "operator": "dx"
	}
	],
	"name" : "Front Door Closed Lights"
}

RezzZ, 2020-03-29:
I really hope these sensors can be added to the list of allowed sensors to control lights.

The control via zigbee sensors should remain in deconz and should not be handled by other homeautomation software like nodered or homeassistant.

An open/close sensor is just as valid as a motion sensor to turn on/off lights. At least for my case:
I have an attic foldable stairs and I have an open/close sensor there. If the stairs are 'open', meaning I can walk up to the attic, I want my lights to turn and stay on. If I close the attic door the lights should turn off.
Same scenario would be valid for e.g. closets.

If the door remains open, the light should be turned on and stay on until the door is closed again...

ebaauw, 2020-03-29:
> I really hope these sensors can be added to the list of allowed sensors to control lights.

What list are you referring to?  I’ve been controlling my lights from Xiaomi Aqara door sensors using deCONZ rules for years.

RezzZ, 2020-03-29:
@ebaauw I'm referring to the deconz frontend.
I can control lights (groups) via switches and sensors
[image]
but magnetic sensors are missing from the list. Switches are there though and they are not senors...

ebaauw, 2020-03-29:
That's Phoscon, which is a separate product and not open source.  No-one here can help with that.

RezzZ, 2020-03-29:
my bad. thought it was part of the rest plugin