dresden-elektronik/deconz-rest-plugin #2147: .71 installer download not on web site?

Ltek, 2019-11-29:
Why is .71 not on deconz.dresden-elektronik.de/win/?C=M;O=D

and, oddly... always wondered why the download links were not posted on the git page... https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Ltek, 2019-12-07:
@manup  ...    .71 has the Osram power-on default, I'd like to use that. Would appreciate if you can post the .71 Windows compiled installer. thx

manup, 2019-12-08:
Hmm I need to check the upload script, something went wrong. The .71 exe is uploaded now.

>and, oddly... always wondered why the download links were not posted on the git page... https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Indeed that would be useful, I'll add them later on.

closed 2020-04-14 by stale[bot]