dresden-elektronik/deconz-rest-plugin #2149: How to reset Ikea led power supply

minscof, 2019-11-30:
Available at (some?) IKEA stores in Berlin. Reset method different for the new ones. They got a dedicated 'Reset' push button. To reset use a pin and push button for about 5 sec.
Didn't had any problem to hook them up to deCONZ, e.g. PHOSCON reading:
```
model TRADFRI transformer 30W
Version 1.2.245
```
[manual pdf](http://www.ikea.com/no/no/manuals/tradfri-driver-for-tradl-s-kontroll__AA-2056402-1_pub.pdf)

_Originally posted by @dxxxm in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/466#issuecomment-372804308_

Smanar, 2019-11-30:
Have you a picture of it ?
On google there is a whole just under the ikea logo.

minscof, 2019-12-01:
There is a very small hole, I didn't see. That's ok now.

closed 2019-12-01 by minscof