dresden-elektronik/deconz-rest-plugin #2152: Looking for a inwall socket/plug

pergolafabio, 2019-11-30:
Hi, I have some Ikea smart plugs, it works good, but all brands smart plugs, just looks like shit ;)

I don't like the extra the adapter, so I am looking for an inwall plug , that is also supported by deconz... Does anyone know one?

Seems xiaomi has some, but they are not for EU market...

https://nl.aliexpress.com/item/32871650876.html?ws_ab_test=searchweb0_0%2Csearchweb201602_0_10130_10152_10151_10618_10059_10696_10084_100031_10083_10547_10624_10546_10623_10307_10868_10548_10869_10341_10065_10340_10068_10343_10342_10103_10620_10344_10325_10545_10622_10324_10621%2Csearchweb201603_0%2CppcSwitch_0&algo_pvid=905b1ddd-db96-437f-bc50-534104486b39&algo_expid=905b1ddd-db96-437f-bc50-534104486b39-12

Thnx

Smanar, 2019-11-30:
Depend where you are from, and features you want, but legrand make EU inwall plug, the gamme is called Netatmo and is available in some EU countries.

pergolafabio, 2019-11-30:
Cool, gonna Google for it.... Don't need an energy meter, but a hardware button for on/off on it is usefull though, so you don't need home assistant to turn it on/off

pergolafabio, 2019-11-30:
Can this one work too? I have everything in house from Niko, so it looks like the rest then...

Not sure if Niko is supported? It's not on compatibility list....

https://www.niko.eu/en/products/niko-home-control/products-for-traditional-wiring/connected-switching-socket-outlet-for-niko-home-control-with-pin-earthing-and-safety-productmodel-niko-aa84a653-36c4-55de-b6d1-d2ddfc6326dc

Smanar, 2019-11-30:
IDK.
On deconz issue, there is only a niko switch.
Nothing on the plug even on other zigbee project.
Nothing on niko website about alexa or hue (that use standard zigbee) and this plug.

SwoopX, 2019-11-30:
Well, the Nikto thingie looks quite interesting, especially since it can handle 16A, measure consumption and has an on-device button (fail safe measure ;)).

I was originally after the ubisys S1/S2, but they are costly and mounted behind the plug...

pergolafabio, 2019-12-02:
> Well, the Nikto thingie looks quite interesting, especially since it can handle 16A, measure consumption and has an on-device button (fail safe measure ;)).
> 
> I was originally after the ubisys S1/S2, but they are costly and mounted behind the plug...

Where do you see that on-device button on that Niko one? I think it's just a led

SwoopX, 2019-12-03:
Description says: "... You can switch the socket outlet on and off locally with the button on the central plate. ..."

pergolafabio, 2019-12-03:
ah cool
gonna visit a store to see, hopefully i can test it and hopefulle it works with deconz
otherwise i return it

SwoopX, 2019-12-03:
I somehow don't think it will work right out of the box but, let's say, you have a very interested person sitting here who might be able to help ;).

ETA is Jan 2020 if I read it correctly in one of the belgian shops...

pergolafabio, 2019-12-03:
Ow , it's not out yet... 
I want to know the price first, Niko is not that cheap... But it's nice to have since all my stuff is from Niko, so it looks similar.... Also those switched looking nice

SwoopX, 2019-12-03:
From what I can tell (googling), it's between 32 - 40€ atm. Fine with me and probably even a better option the the innr plugs.

pergolafabio, 2019-12-03:
thats not exensive indeed for an inwall outlet

Sennevds, 2020-01-21:
Did anyone of you tested the niko outlet with deconz? I'm really interested in buying this one and it looks like you can buy it now https://www.zelektro.be/slim-stopcontact-zigbee-geconnecteerd-schakelbaar.html

pergolafabio, 2020-01-21:
not me :)

pergolafabio, 2020-01-22:
ok, keep us posted!; interested too
maybe you can but it at a local shop, so you can return it if not working

Sennevds, 2020-01-22:
Maybe but I wouldn't know which one will have it. The niko hue remote is nowhere to be found in the standard build markets so I don't think this is going to be in the stores

pergolafabio, 2020-01-22:
if i see it here, ill try it also (belgium)

SwoopX, 2020-01-24:
I'd also buy one if they'd be available. Unfortunately, I haven't seen any technical documentation :/

pergolafabio, 2020-04-01:
some update? :)

Mimiix, 2020-06-05:
As this is not a issue with the software or any other product regarding deCONZ i will close this issue.

closed 2020-06-05 by Mimiix

Sennevds, 2020-06-24:
I have bought one the on off works but electrical measurements aren't exposed to the rest api i've created following issue: #2984

pergolafabio, 2020-06-24:
Ah, thx for sharing, it's already nice that it works with conbee for on/off