dresden-elektronik/deconz-rest-plugin #2153: SyntaxError: Unexpected token u in JSON at position 0

tom75, 2019-11-30:
New Installation of ioBroker, added a few sensors of aquara, was working fine for around 2 weeks without any more changes, since days deconz is no longer working because of this error.

Here is the log:

host.D	2019-11-30 15:45:13.301	error	instance system.adapter.deconz.0 terminated with code 0 (OK)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at emitErrorAndCloseNT (internal/streams/destroy.js:59:3)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at emitErrorNT (internal/streams/destroy.js:91:8)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at Socket.emit (events.js:198:13)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at Socket.socketErrorListener (_http_client.js:392:9)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at ClientRequest.emit (events.js:198:13)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at Request.onRequestError (C:\Program Files\iobroker\SH\node_modules\iobroker.deconz\node_modules\request\request.js:881:8)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at Request.emit (events.js:198:13)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at self.callback (C:\Program Files\iobroker\SH\node_modules\iobroker.deconz\node_modules\request\request.js:185:22)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at Request._callback (C:\Program Files\iobroker\SH\node_modules\iobroker.deconz\main.js:692:28)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: at JSON.parse (<anonymous>)
host.D	2019-11-30 15:45:13.301	error	Caught by controller[0]: SyntaxError: Unexpected token u in JSON at position 0
deconz.0	2019-11-30 15:45:12.777	error	SyntaxError: Unexpected token u in JSON at position 0 at JSON.parse (<anonymous>) at Request._callback (C:\Program Files\iobroker\SH\node_modules\iobroker.deconz\main.js:692:28) at s
deconz.0	2019-11-30 15:45:12.776	error	uncaught exception: Unexpected token u in JSON at position 0

kevdevaux, 2020-01-21:
Samie issue here with deconz.
Versions:

iobroker:
Node.js: v10.18.1
NPM: 6.13.4

deconz:
1.1.2

host.iobroker | 2020-01-21 14:10:56.174 | info | Restart adapter system.adapter.deconz.0 because enabled
-- | -- | -- | --
host.iobroker | 2020-01-21 14:10:56.173 | info | instance system.adapter.deconz.0 terminated with code 0 (NO_ERROR)
host.iobroker | 2020-01-21 14:10:56.172 | error | Caught by controller[0]: at emitErrorAndCloseNT (internal/streams/destroy.js:59:3)
host.iobroker | 2020-01-21 14:10:56.171 | error | Caught by controller[0]: at emitErrorNT (internal/streams/destroy.js:91:8)
host.iobroker | 2020-01-21 14:10:56.170 | error | Caught by controller[0]: at Socket.emit (events.js:198:13)
host.iobroker | 2020-01-21 14:10:56.170 | error | Caught by controller[0]: at Socket.socketErrorListener (_http_client.js:392:9)
host.iobroker | 2020-01-21 14:10:56.169 | error | Caught by controller[0]: at ClientRequest.emit (events.js:198:13)
host.iobroker | 2020-01-21 14:10:56.168 | error | Caught by controller[0]: at Request.onRequestError (/opt/iobroker/node_modules/request/request.js:881:8)
host.iobroker | 2020-01-21 14:10:56.167 | error | Caught by controller[0]: at Request.emit (events.js:198:13)
host.iobroker | 2020-01-21 14:10:56.166 | error | Caught by controller[0]: at self.callback (/opt/iobroker/node_modules/request/request.js:185:22)
host.iobroker | 2020-01-21 14:10:56.165 | error | Caught by controller[0]: at Request._callback (/opt/iobroker/node_modules/iobroker.deconz/main.js:692:28)
host.iobroker | 2020-01-21 14:10:56.164 | error | Caught by controller[0]: at JSON.parse (<anonymous>)
host.iobroker | 2020-01-21 14:10:56.163 | error | Caught by controller[0]: SyntaxError: Unexpected token u in JSON at position 0
deconz.0 | 2020-01-21 14:10:55.622 | info | (18726) Terminated (NO_ERROR): Without reason
deconz.0 | 2020-01-21 14:10:55.618 | info | (18726) terminating
deconz.0 | 2020-01-21 14:10:55.104 | error | (18726) SyntaxError: Unexpected token u in JSON at position 0 at JSON.parse (<anonymous>) at Request._callback (/opt/iobroker/node_modules/iobroker.deconz/main.js:692:28) at self.callback
deconz.0 | 2020-01-21 14:10:55.103 | error | (18726) uncaught exception: Unexpected token u in JSON at position 0
deconz.0 | 2020-01-21 14:10:54.954 | info | (18726) starting. Version 1.1.2 in /opt/iobroker/node_modules/iobroker.deconz, node: v10.18.1
host.iobroker | 2020-01-21 14:10:52.173 | info | instance system.adapter.deconz.0 started with pid 18726
host.iobroker | 2020-01-21 14:10:22.159 | info | Restart adapter system.adapter.deconz.0 because enabled

tom75, 2020-05-20:
well, that is great, closing because no answer on that problem

closed 2020-06-17 by stale[bot]