dresden-elektronik/deconz-rest-plugin #2155: Sengled E13-A21 PAR38 bulb

donnacha, 2019-12-01:
Hi - looking to add device support for the above bulb (Sengled E13-A21 PAR38) with integrated motion sensor. Any help appreciated.

Bulbs do turn on and off via deConz, but no info from motion sensor or ability to toggle motion conditions on or off. 

Clusters:
[image]

Basic cluster:
[image]

Scenes cluster:
[image]

Metering cluster:
[image]

OTAU:
[image]

donnacha, 2019-12-01:
It appears SmartThings implements this product as 'Zigbee Dimmer with Motion Sensor' - see https://github.com/liurenju/buildsys-19-code/blob/0c61fc44f9a04bfa01e43e999cd3718a3d2a8199/smartThingsGroovy/zigbee-dimmer-with-motion-sensor.groovy

ebaauw, 2019-12-01:
According to the `.groovy` file, it has an _IAS Zone_ cluster, presumably for the motion.  Can you try reading the _Simple Descriptor(s)_ in the deCONZ GUI (from the left drop-down button in the node).  After that. the 0x0500 cluster should appear.  If you write the mac address of the RaspBee/ConBee in the _IAS_CIE_Address_ attribute, you should get reports for motion, at least in the GUI.

donnacha, 2019-12-01:
@ebaauw Just tried that - I get a countdown when I reset the node, but no further clusters that I can see.

[image]

[image]

stevewull, 2020-03-12:
I, too, am using the Sengled PAR38 bulb and would like to access the motion and light sensors in Deconz (and through the integration to Home Assistant, which I use for automation).  My cluster info looks similar to donnacha's.

I tried your suggestion: turned on _read simple description_ and was able to see the 0x0500 cluster. I pasted the mac address of the _Conbee II_ into the _IAS_CIE_Address_. No change in the Phoscon App. I do not see the motion and light sensors.

[image]

[image]

SwoopX, 2020-06-04:
Ok, can you read the IAS zone attributes and share a screenshot? In doubt, add the IAS CIE address as you did before and write that attribute, then issue a zone enroll request with the right zone type right above by pressing `exec`. That should trigger enrollment.

Read the IAS zone attributes again afterwards. It should hopefully report enrolled.

closed 2020-07-11 by stale[bot]