dresden-elektronik/deconz-rest-plugin #2156: Unable to start GUI via RDP Xorg

jaddel, 2019-12-01:
When i enter `sudo service deconz-gui status`

i get the following output.

`
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-12-01 12:20:59 CET; 28s ago
  Process: 4189 ExecStart=/usr/bin/deCONZ --http-port=80 (code=exited, status=1/FAILURE)
 Main PID: 4189 (code=exited, status=1/FAILURE)`

Larsn1, 2019-12-01:
Have you stopped the deconz service before?

jaddel, 2019-12-01:
yes

jaddel, 2019-12-01:
i followed this guide to install the desktop since i was coming from a ligth img. 
https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/raspbian-stretch-desktop-und-xrdp-nachinstallieren

Could this be the root of my problem?

closed 2020-04-07 by stale[bot]