dresden-elektronik/deconz-rest-plugin #2157: Trying to POST APi-key returns "invalid Json" Error

bubffm, 2019-12-01:
Trying to connect to REST API via various Rest clients on MacOS without success.

When POSTing http://192.168.x.xx:80/api

i get;

`[
    {
        "error": {
            "address": "/",
            "description": "body contains invalid JSON",
            "type": 2
        }
    }
]`

Any idea?  The IP and port are definitely correct.

bubffm, 2019-12-01:
Solved

closed 2019-12-01 by bubffm

Vissertje1980, 2019-12-11:
I'm struggling with the same problem, what was your solution??

Smanar, 2019-12-11:
What are the data you are posting with this command ?

Vissertje1980, 2019-12-12:
I did someting wrong, i dit not send the body with the request. Yesterday i was able to get an api key :-)

bubffm, 2019-12-15:
only saw this now.  That was actually also my error.  Missing body....