dresden-elektronik/deconz-rest-plugin #2158: Support for Namron 4-channel switch (Luzense/Sunricher clone)

eckhart, 2019-12-01:
This is widely available in Norway, and fits the standard ELKO frames which are common here:

https://www.elektroimportoren.no/namron-zigbee-4-kanaler-bryter-k8/4512703/Product.html

To me it seems more or less identical to the 4-channel ones in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1509 but under a different brand name. It shows up in deCONZ with "Namron AS" as manufacturer instead of "Sunricher" and doesn't show up as a switch in Phoscon/REST API after the switch confirms successful pairing.

Below is what I have from deCONZ. I've read out all the attributes from the seperate channels, they look identical so just including one of them here. Let me know if more info is needed.

[image]
[image]
[image]
[image]
[image]
[image]

corvy, 2019-12-02:
@eckhart it does not show up in groups  in the old web interface either? 

@ebaauw I guess it takes a very basic change to add support for this?

eckhart, 2019-12-02:
No, it doesn't show in the old web interface either.. Would be great to have it supported as it's a lot cheaper than other brands.

ebaauw, 2019-12-02:
Probably.  Just whitelist it in `supportedDevices` in `de_web_plugin.cpp` and assign it to the buttonmap for the 8-button switch in `sensors.cpp`.

eckhart, 2019-12-07:
@ebaauw Sorry that I haven't had the time to follow this up, would be great if no more than this is needed. Being new to this, do the devs prefer that I add, build and test this myself and then make a pull request?

ebaauw, 2019-12-07:
> do the devs prefer that I add, build and test this myself and then make a pull request?

That's what I've done for the devices I own or could borrow.  Adding support without having physical access to the device is almost undoable, due to different interpretations of the ZigBee standards by different manufacturers, and due to the complexity of the REST API plugin code (there's always one more place to whitelist a device).

monsivar, 2020-01-04:
> @ebaauw Sorry that I haven't had the time to follow this up, would be great if no more than this is needed. Being new to this, do the devs prefer that I add, build and test this myself and then make a pull request?

Do you have an config that works for this switch?

eckhart, 2020-01-10:
@monsivar Unfortunately no, I haven't had the time to set up a building and testing environment yet. I'll submit a pull request when I have something working.

monsivar, 2020-01-17:
Is there any recipes on the net for adding this switch to the deconz for an nuub like me? @ebaauw

ebaauw, 2020-01-17:
None official.  I did a writeup here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061#issuecomment-568191933

phithor, 2020-01-23:
I'd like support for this switch myself. Going to try to do this over the weekend.

monsivar, 2020-01-23:
> I'd like support for this switch myself. Going to try to do this over the weekend.

Nice, I did not see the effort of trying..

flemmingss, 2020-01-25:
I am also interested in getting this Namron 4-channel switch to work with Deconz,
I have one still unopened because of this support missing. I am not in any way experienced with doing anything non-phoscon at this time, but if there is some info I can supply just let me know

phithor, 2020-01-27:
I tried to whiteliste it and button map it as described above and then i was unable to build it in my environment, got an error with missing QT module serialport for some reason and did not have time to debug it. Do I need to install qt explicitly in addition to deconz-dev?

https://github.com/dresden-elektronik/deconz-rest-plugin/compare/master...phithor:namronswitch

mk1blacklimited, 2020-01-31:
+1 for wanting this supported! :)

monsivar, 2020-02-04:
> buntu EDIT: Set up a fresh new Ubuntu VM and was able to build it. I re-paired the switch but nothing showed up in Phoscon. Maybe I need to whitelist some more places as linked to above.

Any update on this?

bardnilsen, 2020-02-08:
@eckhart I am having trouble setting the switch in network pairing mode. How did you do it? (Namron 4512703)
I tried holding down on and off buttons in group1 followed by pressing off button in group1 when the led turned on. It blinks 13 times then stops. Any ideas?

lrzorg, 2020-02-09:
+1 for supporting Namron switches.
New to HA and bought some of the 2-channel ones (4512702) thinking any ZigBee 3.0 switch would work :-/
The hardware seems visually equal to the iCasa switches, apart from the button symbols...

About pairing:
Not sure if this helps, but the manual states that if the device is in "network mode" (already paired), it needs to be factory reset before pairing.
The device is in "network mode" if the led blinks when a button is pressed.
To do a factory reset, as per the manual press and hold the first "1" button until the light goes on, then press 5 times on the "1" button. **The 5 presses cannot be too fast in my experience. This took me a couple of tries to get right. About one second between each press seems right**
The device responds by blinking three times if successful.
If pressing a button does not cause the led to blink, it is ready for pairing again.
Pairing mode should be just holding down the first "1" until the light goes on, then press the first "0".

This worked for the 2 channel switch to behave as intended (still needs deconz support), and I checked the manual for the 4-channel and it is the same setup.

monsivar, 2020-02-19:
Any updates on this one?

crazyclona, 2020-03-09:
> Any updates on this one?

I hope there is soon ;) Would be awsome.. bought a few of them.......

phithor, 2020-03-09:
OK guys, I have not figured out if my branch works, I think I need to try again. Or can any of you help test my branch above? Build it, install it, test of you can add the button.

bardnilsen, 2020-03-10:
Hi phithor,

Great work :)

I have tested it. I can now add the 4-channel switch in phoscon app, but it does not show up in phoscon web gui. I see it in the rest-api and shows up in my HomeAssistant, but with no entities.

phithor, 2020-03-13:
Ah, thanks for that, I did not understand what worked and what not. At least we have gotten somewhere when it shows up in the rest-api. It's corona-time, so I think I will have time to do some more on this today/this weekend.

phithor, 2020-03-14:
Hah, in conclusion the commit I made almost two months ago works perfectly fine. 
I did it the same as was done here: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/d5554bf5e47f96a609fb40df7eb96c047907a5aa

The key to fixing this is pressing open network and reding the basic cluster of each of the groups while the network is open. Then the groups will be made where you can add lights to. 
See info in this thread:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1509#issuecomment-555191920

I will push the commit to the main repo asap.

monsivar, 2020-03-24:
@phithor  cloned your git https://github.com/phithor/docker-deconz

Then I get this problem when trying to test, after running the command: qmake && make -j2

In file included from bindings.cpp:11:
de_web_plugin.h:16:10: fatal error: deconz.h: Ingen slik fil eller filkatalog
 #include <deconz.h>
          ^~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:527: release/bindings.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from authorisation.cpp:13:
de_web_plugin_private.h:23:10: fatal error: sqlite3.h: Ingen slik fil eller filkatalog
 #include <sqlite3.h>
          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:509: release/authorisation.o] Error 1
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'
make: *** [Makefile:40: release] Error 2

phithor, 2020-03-24:
Hm, I think I got that as well the first time, but I believe that is missing files in your qt setup? Have you install deconz-dev?

monsivar, 2020-03-24:
> Hm, I think I got that as well the first time, but I believe that is missing files in your qt setup? Have you install deconz-dev?

I have. Restarted my pi, and its working :-)

monsivar, 2020-03-25:
> Hm, I think I got that as well the first time, but I believe that is missing files in your qt setup? Have you installed deconz-dev?

Hi. but i still have an problem . The switch is not listed upp in the old web interface, or in the api.

[image]

phithor, 2020-03-25:
Did you follow the instructions I linked to above? In the old interface, open the network and then read cluster in each of the channels. That should create the groups.

monsivar, 2020-03-25:
I did, but no new switch. Was also thinking about removing the switch first, then including it, but did not find any way of doing that. Is that needed?

phithor, 2020-03-25:
You will not see a new switch. Only a new group in the old web interface. See the info i linked to above, there you see the end result as well.

monsivar, 2020-03-25:
> You will not see a new switch. Only new groups in the old web interface. See the info i linked to above, there you see the end result as well.

No new group either. I have done all that is listed in the linked info, but no results

phithor, 2020-03-26:
Hi, I tested it again with a fresh version without the groups and it successfully created the groups again. Took a few minutes so let it work after you have read the basic cluster of each channel. So I believe it should work. 

I have bound one of the on/off clusters to a light on/off in deCONZ GUI , does that make it not create a group for that channel? I only got three this time, and tested it twice. 

[image]

monsivar, 2020-03-26:
Tested again from the beginning.

1: git clone https://github.com/phithor/deconz-rest-plugin
2: cd deconz-rest-plugin
3: qmake && make -j2
4: sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
5: confirm updatet libde_rest_plugin.so
5: started DeCONZ
6: Open network (10min)
7: Cluster info read an push switch on all 4 Basic (i see that DeCONZ is getting data
8: waiting for several minuts, an still no new groups in the old web interface.

[image]

[image]

[image]

corvy, 2020-03-26:
Might be related to the problems... I have the Namron verions, two of them. Due to some noise from Wi-Fi I have tried to move my Zigbee network off channel 11 onto channel 20 or 25 (tried these two, not 15). On the two I tried the switches both stopped working. Tried a ferw times but no success. They only seem to work on channel 11.  

Also, remember to use a new battery, that was my problem earlier on.

phithor, 2020-03-26:
Ah, you have to pull the namronswitch branch from my repo, not the master branch ! ;)

monsivar, 2020-03-26:
> To check if you are on the right version you will see Software version 2.05.75 on the REST-API as I have merged in all recent changes.

Ok, i now did:  git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git
That is the right one?

This is the old web gui result now
[image]

> Also, remember to use a new battery, that was my problem earlier on.

The switch is responding in web gui, so i dont think that is the problem

phithor, 2020-03-26:
and then "git checkout namronswitch", then you will have the correct content in the files

monsivar, 2020-03-26:
Aha. thanks for all the help. Did not get that point, first time workin on this

1: git clone https://github.com/phithor/deconz-rest-plugin
2: cd deconz-rest-plugin
3: git status
4: git checkout -b namronswitch
3: qmake && make -j2
4: sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

crazyclona, 2020-05-10:
Anyone know if this patch will be included in thee official branch?

pselnes, 2020-05-26:
This should be included in the last version v76/77.

I managed to create the groups by reading the cluster info as described above.
However, the switch seems to go sleep after a short while? The first keypress is ignored (no light on the switch), but on a consecutive press, the switch lights up and the lights respond.

Is this the case for everyone?

crazyclona, 2020-05-31:
Seems it works better now. though, I for some reason do not get anything on the api..

flemmingss, 2020-07-05:
Deconz 2.05.78 /22.05.2020
Firmware 26330500

Phoscon gw showing "Update to 26350500", but it looks like it is not updating, even if it looks like it is in the interface.

Should it work for me on this version? I tried to add the Namron 4-channel switch yesterday by the Phoscon GUI, but I did not find any device.

phithor, 2020-07-05:
You will not be able to add it through the Phoscon GUI, you need to go to the old interface and press open network, then open deconz gui and read cluster for all 4 groups of the namron switch. Then you will see the groups. Read above for more info.

habakke, 2020-07-09:
I am having the issues getting this setup working, only seeing the 8 ch switch (Namron) battery level status in Home Assistant and not having any Groups created in the old-webapp to which I can assign devices. The setup is running the latest deConz 2.05.78 release on a Conbee II with 26580700 firmware.

[image]

In the old-webapp, no groups are created for the switch when following the procedure below:
1. Reset device
2. In the Phoscon web-app, choose to search for new switch (Philips or Other)
3. Follow devices procedure for "Zigbee Network Pairing through Coordinator or Hub"
4. Read all of the 4 basic cluster groups in deConz UI.

[image]

The switch does show up in the deConz UI:

[image]

Any idea how to get the groups created so I can assign devices to be controlled by the switch? And to have additional entities show up (besides the battery level) in Home Assistant through the integration? I have tried to re-run the procedure many times, and reading the basic cluster groups multiple time, but still the same result.

phithor, 2020-07-09:
You need to press "Open Network" in the old web-app. Then re-read the cluster info. There is nothing else to it. I am not sure what other tricks that can be used, but remember to set the open network to enough time.

habakke, 2020-07-10:
> You need to press "Open Network" in the old web-app. Then re-read the cluster info. There is nothing else to it. I am not sure what other tricks that can be used, but remember to set the open network to enough time.

I did a test again. Before starting I did a reset of the gateway and rest of the Namron switch.

1. Changed the duration the network is kept open in the old web-app:

[image]

2. Opened the network in the old web-app

3. While the network was open, I executed the "Zigbee Network Pairing through Coordinator or Hub" procedure to pair the switch with the gateway. (Essentially holding down the first pair of on/off buttons until the light come on and the immediately pressing the first off button) It then blinks to indicate pairing is successful and I can se the device showing up in the deConz UI.

4. While the network is still open, manually read all the cluster groups using the deConz UI. (Clicking on the group, selecting the "Cluster info" tab, pressing the read button and then a button on the switch to wake it up.

Still the same result as before, no groups are created and visible in the old web-app. Am I doing something incorrect?

Mimiix, 2020-07-10:
@habakke It could be possible that the device is not in Phoscon  yet. They need to add it manually. However, the RestAPI (deCONZ) does show them thus they are implemented correctly. Phoscon is closed source, so we can't do that.

If you need it implemented, please open a git issue https://github.com/dresden-elektronik/phoscon-app-beta

I'll be closing this issue as it seems to be resolved and implemented.

SwoopX, 2020-07-10:
It is implemented, but not properly. There's some afterwork to be done, mainly around the groups and binding. I can see if I can fix it once I have the new switches to oversee all the implications.

closed 2020-07-10 by SwoopX

habakke, 2020-07-10:
@Mimiix && @SwoopX I am sort of confused whether this is added correctly or not? It is definitely not working here.

For good measure I created a git issue as suggested by @Mimiix: https://github.com/dresden-elektronik/phoscon-app-beta/issues/152

SwoopX, 2020-07-10:
It is supported and also whitelisted in the code. However, the support was provided by someone who obviously wasn't that familiar with the code.

As such, the underlying request is done. Visibility in Phoscon is a different thing, as Mimiix pointed out, therefore the requirement for a request.

phithor, 2020-07-11:
@habakke i checked your issue in the Phoscon App and I noticed your switch shows manufacturer id "NAMRON AS" in all caps. I have not seen that before. That might why it is not working for you, as the whitelist is only for "Namron AS" if I remember correctly. 
Can you confirm the manufacturer ID shows all caps?

habakke, 2020-07-12:
@phithor I can confirm the manufacturer name show all caps "NAMRON AS" as seen in the screenshot of the basic cluster group below:

[image]

The OTAU cluster group shows the manufacturer ID to be 0x1224 as seen below:

[image]

habakke, 2020-07-12:
@phithor I installed the dev env. and did a quick change to the code to assign the sunricher buttonmap if the manufacturer matches upper case "NAMRON AS", compiled and installed. Now the groups are created as expected:

[image]

SwoopX, 2020-07-13:
Taken care of this in my last PR real quick. While pairing, we now set the Manufacturer to "Namron AS" for all devices beginning with 451270. Should be included in the next release.

clindland, 2020-07-21:
> Taken care of this in my last PR real quick. While pairing, we now set the Manufacturer to "Namron AS" for all devices beginning with 451270. Should be included in the next release.

Mine still won´t work. After this release I checked the model identifier number, and it says 4512721, and still "NAMRON AS" with all caps. Is it possible to include the switches with these id numbers?

[image]

SwoopX, 2020-07-21:
Renaming will be done based on the modelId. Of course your device brings another one into play, which doesn't fit the pattern... How many different IDs can one device have...?

clindland, 2020-07-28:
Yeah, they didn't make it easy on us.. Is it possible to add my ID's (or my pattern if there's anyone out there with an ID similar to mine) to the renaming pool? It would be really great to get this to the deconz hassio addon 👍

SwoopX, 2020-07-28:
Don't worry, will be supported with the next release.

flemmingss, 2020-08-03:
Do you know when that is? because I am on the least one now and am not able to discover it.

[image]

SwoopX, 2020-08-03:
August 15th, if you run bare Raspian, without HA and have subscribed to the beta channel.

[11 more replies]