dresden-elektronik/deconz-rest-plugin #2161: Deconz and xiaomi door sensor - multiple messages

Malekith, 2019-12-01:
Hi,
Not sure is someone can help me with this, but I have deconz set up with Home assistant, using the Conbee2. 
For some reason when I open a door with the door sensor, it will send multiple status change messages. 
For example, it will send Open, Closed, Open, when I open the door and then via versa on close. 

This is fine as the final status is correct. However sometimes it send one extra and then is set to the incorrect status. 

Any ideas?

Thanks.

ebaauw, 2019-12-01:
Double-check the sensor battery.  If it's under 100%, replace it and check whether the problem still occurs.  If so, double check with `dc_eventlog` if deCONZ sends multiple websocket notifications or if Home Assistant is imagining them.

Malekith, 2019-12-01:
Hi ebaauw,

I have already replaced one of the batteries and it is still doing happening. 

How do I check the dc_eventlog? Where is it’s location?

Thanks again.

hellcry37, 2019-12-08:
using nuc / home assistant / ubuntu 18 lts / deconz in docker and i do not have this issue on 3 xiaomi aqara open-close sensors.

what sensors do you use, mi or aqara?

ebaauw, 2019-12-08:
> How do I check the dc_eventlog?

It's a command line utility to log the web socket notifcations, included in homebridge-hue.

> what sensors do you use, mi or aqara?

That's a very good question.  I only have Aqara sensors (the square-ish models); no Mi sensors (the round-ish models).  I've seen various reports of issues with the Mi sensors; my Aqara door, weather, and leak sensors, as well as the wireless switch/button only ever acted weird on low battery.  As did the Trådfri motion sensor (also using CR2032 button cell batteries).

Malekith, 2019-12-09:
Sorry for the late reply. Been away for work. 

I have the aqara square ones. 

I’m running on a raspberry pi | raspbian | hassio docker | deconz integration. 

I’ll check the logs tonight and get back to you. 

Still having the issues.

Malekith, 2019-12-13:
Sorry but can you point me in the direction of the dc_eventlog?

i cant seem to find where it is located.

Thanks,

ebaauw, 2019-12-13:
On installing homebridge-hue, a symlink should be created in `/usr/local/bin` or `/usr/bin`, depending on how you installed NodeJS.  The utility itself is installed to `/usr/local/lib/node_modules/homebridge-hue/cli` or `/usr/lib/node_modules/homebridge-hue/cli`.

Malekith, 2020-01-02:
Ok, sorry again for the late reply.

i have had a look and i am unable to locate this log file.

Just for some information. I am running Hassio in a docker on Raspbian.

Deconz is installed via the hassio add on.

Not sure where to locate this log file.

closed 2020-05-09 by stale[bot]