dresden-elektronik/deconz-rest-plugin #2162: Need more info from Eurotronic Spirit Zigbee

broekieman, 2019-12-02:
Hi,

I bought my Spirit Zigbee Thermostat and want to create a script that gives me the status of the valve position. Is this possible?

Smanar, 2019-12-02:
Sure. You have the valve state on the JSON.
This one is older but it's an exemple
{
  "config": {
    "battery": 100,
    "boost": false,
    "displayflipped": true,
    "heatsetpoint": 2100,
    "locked": false,
    "off": false,
    "offset": 0,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "19c89536ce4a0af7399c4405f78e516d",
  "manufacturername": "Eurotronic",
  "modelid": "SPZB0001",
  "name": "Living Room Radiator",
  "state": {
    "lastupdated": "2019-02-10T14:54:26",
    "on": true,
    "temperature": 2309,
    "valve": 82
  },
  "swversion": "15181120",
  "type": "ZHAThermostat",
  "uniqueid": "00:15:8d:00:01:92:d2:51-01-0201"
}

You have a "valve" value (in % I think), and the "on" state.

broekieman, 2019-12-02:
I am not an experienced user, but a Quick learner. How can I use this in Domoticz? I want to have a lua script using the OFF and BOOST to open and close the valve. It must check the position of the valve before any action is taken.

Thanks in advance.

Smanar, 2019-12-02:
Ok so bad luck for you ^^. I m the dev of the domoticz plugin and ATM I m not using the valve return state.
But i don't see why you want to know the valve state ? You can send a "off" order even the valve is off.
And boost mode is a "special" mode, used in emergency situation, the auto mode is better.

I think you are trying to use an external temperature sensor to regulate the valve ? You are not the first one who are trying, and I don't know someone who has succeeded.

broekieman, 2019-12-02:
So, if I send command to close valve can I be 100% sure the command reaches the valve? Is there a way to get a message when command is not executed?

ebaauw, 2019-12-02:
The REST API doesn’t support setting the valve position, you can only do this from the GUI.  To script this, use https://github.com/ma-ca/deconz-cli-plugin.

In the REST API, there’s no way to be 100% sure that the TRV has executed a command, other than waiting for the next report and checking whether the state has been changed.  This is exactly why the REST API only supports using the TRV’s PID algorithm (where the TRV sets the valve position based on the heat setpoint and current temperature).

When using the cli plugin, you can read the attribute immediately after sending the command.

Smanar, 2019-12-02:
Good question, I haven't take a look on the code for this part, but I think the answer is not. You can have a websocket return saying the mode is changed, but without the device receive the command. On lot of situations, deconz doesn't check the request return. To be sure you can make a try, remove battery on a device, send it a command, and take a look on websocket return (just enable log to see them)

But not a big problem for me, because
- it's rare.
- if the device is realy out of order, deconz will inform you.
- you have a reporting or a pooling. like this one, from another user.

>2019-11-08 18:09:34.105 Status: (deconz) Thermostat debug : {'uniqueid': '0201', 'id': '85', 't': 'event', 'state': {'on': True, 'valve': 17, 'lastupdated': '2019-11-08T17:09:34', 'temperature': 2105}, 'r': 'sensors', 'e': 'changed'}
2019-11-08 18:12:15.795 Status: (deconz) Thermostat debug : {'uniqueid': 0201', 'id': '85', 't': 'event', 'state': {'on': True, 'valve': 34, 'lastupdated': '2019-11-08T17:12:15', 'temperature': 2105}, 'r': 'sensors', 'e': 'changed'}
2019-11-08 18:14:16.156 Status: (deconz) Thermostat debug : {'uniqueid': 0201', 'id': '85', 't': 'event', 'state': {'on': True, 'valve': 24, 'lastupdated': '2019-11-08T17:14:16', 'temperature': 2105}, 'r': 'sensors', 'e': 'changed'}
2019-11-08 18:29:36.595 Status: (deconz) Thermostat debug : 

Deconz send itself informations everytime something is changing, or after a time period. So even you have a missed a command, it will be corrected some time after.

For exemple, you send the "off" mode, you receive a "fake" return to confirm it, domoticz will set the device to "off", but if 10mn after you have a return with "mode" = "auto", domoticz will change it too. So your LUA script can trigger again and send the 'off' command again.

broekieman, 2019-12-02:
Thanks to you both, know more about Deconz and how it works now. At the moment I can only create a script with blocky and it's going the way I like. Why is column Data not updated to OFF? Btw I use my valve to close access to floorheating so I can set the burner to a higher temperature to raise temperature in the bathroom. 
[image]

[image]

Smanar, 2019-12-03:
Right, an empty field as Vloerverwrmaing aant/uit is not normal.
You can try to set manually the boost or an other mode directly on the device, the value need to change too on domoticz.

But I don't see problem on your blocky code, have you enable logging, and take a look on logs ?

BTW, if you still have problem, better to make an issue on domoticz or plugin github, it's more a domoticz problem than a deconz one. I m on both.

closed 2020-04-08 by stale[bot]