dresden-elektronik/deconz-rest-plugin #2163: IKEA Fyrtur Blinds unresponsive after Raspberry reboot

Larsn1, 2019-12-02:
I am using the latest version of Deconz 2.05.71 and homebridge-hue 0.11.44  with rasbee (old version) and tried to pair the blinds this weekend as I finally got permission from the gov. to buy one :)

Pairing of the blinds is going well, but as soon I fully restarted the raspberry to get them into homekit they are unresponsive (firmware version disapears as well) and don't react in the home app nor in the phoscon web interface.

I have paired them one more time and just restarted homebridge in my last try and now they work. Logs will be uploaded later on.

Moreover there should be a way to pair and configure the up/down remote in the phoscon app. For know I only see it in the old webinterface.

ebaauw, 2019-12-02:
I don’t think Phoscon yet supports the remote.  homebridge-hue does, so you should see it in HomeKit.

Best include the blind in the remote’s group, so you can control it without deCONZ running as well.

My blind occasionally gets kicked out by its parent Hue light, and somehow doesn’t find another parent.  Symptoms: the blind seems unresponsive to commands, but deCONZ is still receiving attribute reports.  Workaround: open the network/search for devices and briefly disconnect the battery from the blind.  Don’t reset it, or you need to redo the calibration and the group assignment.   I’ve yet to try and include the Trådfri repeater in my production network.

Larsn1, 2020-01-01:
Thank you Eric and a happy new year.
I had to remove the battery several times so far and have gone through the procedure as described. Today I noticed that the fyrtur blinds become unresponsive after you use the physical buttons on the blinds.

For background, I don’t use the remote but Siri for everything and didn’t had my phone with me. Therefore I opened the blinds by pressing the small open button next to the battery.

Another way to make them unresponsive is to unplug and replug the assigned repeater. My wife needed the power socket for the vacuum cleaner....
Is that something that can be fixed or is the procedure you described the only way?

ebaauw, 2020-01-01:
> Today I noticed that the fyrtur blinds become unresponsive after you use the physical buttons on the blinds.

Did it become unresponsive, or did you only notice it at that time?  Did the gateway receive notifications from the blind (for the changing position) when you pressed the button?

> Another way to make them unresponsive is to unplug and replug the assigned repeater.

Yeah, you really don't want to power down ZigBee routers.  Even so, the Fyrtur should find a new parent, when its former parent goes missing.  Same question: does the gateway still receive periodic reports or has the blind gone incommunicado altogether?

> Is that something that can be fixed or is the procedure you described the only way?

Not until we figure out what happens exactly.  Even then, if this is between the Fyrtur and its parent router, there's very little deCONZ or the RaspBee/ConBee can do about it.

I did add the Trådfri repeater to my production network, and forced the blind to select it as parent (by powering down all other routers in range, while resetting and re-pairing the blind).  I'm pretty sure the repeater had been powered all the time, but even then, after a couple of days, gateway commands would no longer reach the blind.  Not seeing any improvement, I then removed the repeater.

closed 2020-05-07 by stale[bot]