dresden-elektronik/deconz-rest-plugin #2168: SPZB0001 turns unreachable in the REST API despite being connected and reachable in the deCONZ GUI

ginkel, 2019-12-04:
I've got multiple SPZB0001 thermostats in operation, connected to a ConBee II via various routers (IKEA TRADFRI, innr SP120). One of them (connected via a TRADFRI bulb) regularly shows up as `reachable: false` in the REST API despite being shown as connected and reachable (and I can read/write clusters) in the deCONZ GUI:

```
  "29": {
    "config": {
      "battery": 100,
      "displayflipped": null,
      "heatsetpoint": 1600,
      "locked": null,
      "mode": "auto",
      "offset": 0,
      "on": true,
      "reachable": false
    },
    "ep": 1,
    "etag": "2458aa6174f2d3ca13d8ba350575cd31",
    "manufacturername": "Eurotronic",
    "modelid": "SPZB0001",
    "name": "SPZB0001",
    "state": {
      "lastupdated": "2019-12-03T05:32:13",
      "on": false,
      "temperature": 1900,
      "valve": 0
    },
    "swversion": "20191014",
    "type": "ZHAThermostat",
    "uniqueid": "00:15:8d:00:01:ff:c4:3c-01-0201"
  },
```

Any ideas how to debug this? Restating the deCONZ Docker container usually solves the issue, but I'll leave the current state as is for a while in case you'd like me to collect some information while the issue is still current.

The logs repeatedly show such fragments - those being the last ones referencing 0x00158D0001FFC43C:

```
2019-12-03T20:23:59.272767662Z 21:23:58:081 delay sending request 174 dt 4 ms to 0x00158D0001FFC43C, cluster 0x0201
2019-12-03T20:23:59.272912271Z 21:23:58:181 delay sending request 174 dt 4 ms to 0x00158D0001FFC43C, cluster 0x0201
2019-12-03T20:23:59.273056324Z 21:23:58:228 delay sending request 174 dt 4 ms to 0x00158D0001FFC43C, cluster 0x0201
```

Any ideas?

dedepene, 2019-12-04:
Exact same problem here! I have 3 of those and one is acting up.

```
18:38:11:178 delay sending request 238 dt 4 ms to 0x00158D000192DC31, cluster 0x0201
18:38:11:278 delay sending request 238 dt 4 ms to 0x00158D000192DC31, cluster 0x0201
18:38:11:378 delay sending request 238 dt 4 ms to 0x00158D000192DC31, cluster 0x0201
18:38:11:478 delay sending request 238 dt 4 ms to 0x00158D000192DC31, cluster 0x0201
18:38:11:578 delay sending request 238 dt 4 ms to 0x00158D000192DC31, cluster 0x0201
18:38:11:678 delay sending request 238 dt 4 ms to 0x00158D000192DC31, cluster 0x0201
18:38:11:778 delay sending request 238 dt 4 ms to 0x00158D000192DC31, cluster 0x0201
18:38:22:911 0x00158D000192DC31 error APSDE-DATA.confirm: 0xA7 on task
18:38:27:957 0x00158D000192DC31 error APSDE-DATA.confirm: 0xA7 on task
```

Here #10 is the one with issues while #11 is one that works fine:

```
"10": {
        "config": {
            "battery": 100,
            "displayflipped": null,
            "heatsetpoint": 1750,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "6cbb62488aa941bd4536fc89baa55755",
        "manufacturername": "Eurotronic",
        "modelid": "SPZB0001",
        "name": "SPZB0001",
        "state": {
            "lastupdated": "2019-12-04T16:22:46",
            "on": false,
            "temperature": 1990,
            "valve": 0
        },
        "swversion": "20190408",
        "type": "ZHAThermostat",
        "uniqueid": "00:15:8d:00:01:92:dc:31-01-0201"
    },
    "11": {
        "config": {
            "battery": 95,
            "displayflipped": null,
            "heatsetpoint": 2200,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "b23ca5386f7e9e582a3eebe13ab2e8c0",
        "manufacturername": "Eurotronic",
        "modelid": "SPZB0001",
        "name": "SPZB0001 (2)",
        "state": {
            "lastupdated": "2019-12-04T16:38:03",
            "on": true,
            "temperature": 2273,
            "valve": 187
        },
        "swversion": "20190408",
        "type": "ZHAThermostat",
        "uniqueid": "00:15:8d:00:01:92:db:bc-01-0201"
    },
```

ginkel, 2019-12-04:
I have browsed the source for a while, but I can't seem to find a place where a sensor is set back to "reachable" after it has become unreachable. That's different for lights, though.

Smanar, 2019-12-04:
IDK if it's that you are searching but
- there is a checkSensorNodeReachable function in de_web_plugin.cpp
- there is some code in database.cpp around line 3497

ginkel, 2019-12-04:
Probably ;-)

One interesting discovery. Just recreated the Docker container to enable debug logging. That didn't bring the device back to "reachable". The device in question shows up ad neighbor, though:

```
2019-12-04T17:25:49.445586716Z 18:25:40:832 ZDP status = 0x00 -> SUCCESS
2019-12-04T17:25:49.445609345Z 18:25:40:832 ZDP Mgmt_Lqi_rsp zdpSeq: 42 from 0x000B57FFFE877CA7 total: 14, startIndex: 12, listCount: 2
2019-12-04T17:25:49.445632863Z 18:25:40:832     * neighbor: 0x00158D0001FFB10C (0x3D11), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
2019-12-04T17:25:49.445654993Z 18:25:40:832     * neighbor: 0x00158D0001FFC43C (0x17BE), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
```

dedepene, 2019-12-05:
Coincidentally, I had a Xiaomi T&H sensor that got disconnected at about the time the TRV started acting up. I also noticed other devices struggling along. As soon as I re-paired the xiaomi sensor and everything is back to normal. How does this make sense, provided the xiaomi device is not even a router for other end devices (like the TRV) to rely on....?

```
"10": {
        "config": {
            "battery": 100,
            "displayflipped": null,
            "heatsetpoint": 1900,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            **"reachable": true**
        },
        "ep": 1,
        "etag": "ab5c0217ca70bedd1dc8920c532509f4",
        "manufacturername": "Eurotronic",
        "modelid": "SPZB0001",
        "name": "SPZB0001",
        "state": {
            "lastupdated": "2019-12-05T07:57:47",
            "on": false,
            "temperature": 2007,
            "valve": 0
        },
        "swversion": "20190408",
        "type": "ZHAThermostat",
        "uniqueid": "00:15:8d:00:01:92:dc:31-01-0201"
    },
```

ginkel, 2019-12-05:
I also have a seconds device that is unreachable, but for a very good reason. I had to request an exchange, because it was defective:

```
  "30": {
    "config": {
      "battery": 100,
      "displayflipped": false,
      "heatsetpoint": 1600,
      "locked": null,
      "mode": "auto",
      "offset": 100,
      "on": true,
      "reachable": false
    },
    "ep": 1,
    "etag": "1bef868313aa25a096b03ed179f6c91a",
    "manufacturername": "Eurotronic",
    "modelid": "SPZB0001",
    "name": "SPZB0001 (2)",
    "state": {
      "lastupdated": "2019-12-01T11:34:13",
      "on": false,
      "temperature": 2000,
      "valve": 0
    },
    "swversion": "20191014",
    "type": "ZHAThermostat",
    "uniqueid": "00:15:8d:00:01:ff:ac:45-01-0201"
  },
```

Is there a way to get rid of it? I already deleted it from the deCONZ GUI twice, but it remains in the REST API and also keeps showing up again (despite disconnected) in the deCONZ GUI after a while.

ebaauw, 2020-03-15:
`reachable` is utterly useless and best ignored, see #2590.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix