dresden-elektronik/deconz-rest-plugin #2169: no api key required for websocket

mqtt-fan, 2019-12-04:
Hi all,

I tried the java script sample from the deconz rest documentation.

It works very well, but the websocket can access the device without any authentication. 
How can I acitvate authentication?

Smanar, 2019-12-05:
The websocket don't acess to device. It just permit deconz send notifications in real time for other applications. You can't make action on device with it.

But yes, ATM there is no authentification for it, so someone can connect to it to see the temperature in your living room, or if you have turn off the light.

johanneswilm, 2019-12-09:
@Smanar That sounds like a really bad security issue. Is this open source so that others can fix it if Dresden Elektronik doesn't have the resources to do so?

Smanar, 2019-12-09:
It depends on the point of view.

I was joking on my previous post ^^. For me using https is totaly useless too. The only security issue I can find is if someone use a door locker and using the api url with random apikey and it depend of your own security network, easy to securise.
You can publish your websocket on live streaming if you want, no danger for you.

I don't know their priorities, but for me there is lot of things more important than theses security issues.

closed 2020-04-14 by stale[bot]