dresden-elektronik/deconz-rest-plugin #2170: Odroid N2 ARM64

fle44, 2019-12-05:
Hi,

I'm running my smart home from an arm64 os system (the Odroid N2) with ubuntu 18 and with a soft called jeedom, I bought a conbee 2 USB and some zigbee sensors.

I just realize, as I'm trying to install it, that there is no arm64 deCONZ binary available.

Is there a way to run deCONZ on an arm64 system and to make it compatible with my software jeedom?

Or better have you planed to develop a arm64 library even in beta ?

Best regards

Thanks,

Franck

manup, 2019-12-05:
There are .deb packages available for arm64.

http://deconz.dresden-elektronik.de/raspbian/alpha/

These can be installed as described briefly in:
https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages-for-manual-installation

fle44, 2019-12-07:
> Message du 05/12/19 19:03> De : "Manuel Pietschmann" > A : "dresden-elektronik/deconz-rest-plugin" > Copie à : "fle44" , "Author" > Objet : Re: [dresden-elektronik/deconz-rest-plugin] Odroid N2 ARM64  (#2170)> >
> There are .deb packages available for arm64.
> http://deconz.dresden-elektronik.de/raspbian/alpha/
> These can be installed as described briefly in:> https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages-for-manual-installation

Hello
 
Thank you, it works on debian strech but not with ubuntu 18...
 
Do you have a package for ubuntu 18 arm64 ?
 
best regards
 
Franck

closed 2020-04-12 by stale[bot]