dresden-elektronik/deconz-rest-plugin #2173: missing ubuntu beta deconz-2.05.71-qt5.deb

pvizeli, 2019-12-06:
The deconz-2.05.71-qt5.deb release is missing now on /ubuntu/beta/deconz-2.05.71-qt5.deb.
We can't build our Hass.io Add-on anymore. Also, the release page looks special for that release. Are there somethings they we need known or was this version removed?

manup, 2019-12-06:
Uff, damn you're right, it should have been there. I'll check it out now so it should be up again very soon. Do you mean the https://github.com/dresden-elektronik/deconz-rest-plugin/releases page? There was no change, but we have a new additional changelog page at https://phoscon.de/en/changelog

The old changelog link on dresden-elektronik.de isn't working anymore, unfortunately the redirect isn't in place yet.

~~https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/change-log-von-deconz-und-der-webapp/?L=1~~

manup, 2019-12-06:
The Ubuntu release is back online now :) the final copy step wasn't executed in our build pipeline.

closed 2019-12-06 by pvizeli