dresden-elektronik/deconz-rest-plugin #2175: error APSDE-DATA.confirm: 0xA7 on task

jonskoglund, 2019-12-06:
I have a small Zigbee mesh (Deconz + ConbeeII) with a Hue light that is responding with a serious delay or not responding at all to changes by flipping switch in HA.
The Deconz log shows this:
```
13:54:22:018 device state timeout ignored in state 3
13:54:23:069 device state timeout ignored in state 3
13:54:25:770 device state timeout ignored in state 3
13:54:26:742 timeout on poll APS confirm
13:54:26:762 Current channel 25
13:54:26:772 Device TTL 4500 s flags: 0x7
13:54:27:114 0x0017880102372BAD error APSDE-DATA.confirm: 0xA7 on task
13:54:28:071 device state timeout ignored in state 3
13:54:36:242 device state timeout ignored in state 3
```
Deconz 3.9
ConbeeII FW 264A0700 2.05.71
Attached an image of the mesh.
[image]
Any tips on improving the performance of this mesh are also appreciated :-)

Smanar, 2019-12-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log
Have you try the USB cordon extension, to improve connexion ? (even your network look fine for me, BTW enabling LQI is realy cpu intensive, so enable it just for info or take screenshoot)

jonskoglund, 2019-12-07:
Yes I’m using a usb extension cord. I’m aware of the Error Code page.  But no idea how to fix it. Thanks for the LQI tip.

MikaelHoogen, 2019-12-17:
@jonskoglund I have the same issue. In my case I can control the lights if I make a Deconz group and make group commands. But if I try to control the light by sending a command directly it fails. Could you make a test and see if the same correspond to your setup?

jonskoglund, 2019-12-17:
Yes. I will try this soon. I will share my findings here.

MikaelHoogen, 2019-12-17:
I saw another strange behavour. When I unplugged the device the node was blinking blue in Deconz when I did send an attribute read. It wasn't marked as a zombie and the behavour seem like it actually was responding.

MikaelHoogen, 2019-12-25:
@jonskoglund any updates on the issue?

jan-dewit, 2020-03-11:
@jonskoglund any updates on the issue?

jonskoglund, 2020-03-14:
nope.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix