dresden-elektronik/deconz-rest-plugin #2176: Senic Friends of Hue Smart Switch - no deconz_event ?

bedaes, 2019-12-06:
I just received some Senic Friends of Hue Smart Switches. I succesfully added them in Phoscon App. It's deteced as:

Vendor: PhilipsFoH
Product: Friends of Hue Switch

However, in appdaemon I don't receive any `deconz_event` events when pressing the buttons. When I run `deCONZ --dbg-info=1`and press any buttons there is no output. I would have expected something like `18:17:31:055 button 1002 Toggle` like on my IKEA remotes?

bedaes, 2019-12-07:
Nevermind, I missed step 5 from here: https://phoscon.de/en/support#pairing-friends-of-hue-switch

closed 2019-12-07 by bedaes

sildehoop, 2020-02-10:
hey I have the same problem, but i did do step 5 is it necessary to have a bulb in your house? because i can't put the switch in a group.

sildehoop, 2020-02-10:
[image]

sildehoop, 2020-02-10:
[image]

sildehoop, 2020-02-10:
[image]