dresden-elektronik/deconz-rest-plugin #2177: pwa not running on https, not installable

johanneswilm, 2019-12-06:
Hey,
as a web developer myself I am wondering why the PWA app doesn't have all those things enabled that makes chrome decide to make it installable (such as https, service worker). That would make it considerable better integrated at least with an Ubuntu desktop and Android phones (and probably also other platforms) and it's not that difficult to fulfill the requirements. I'm sure there are reasons for that not having happened, but I didn't see an open issue about this.

mqtt-fan, 2019-12-09:
The Websocket is running on 443

johanneswilm, 2019-12-09:
@mqtt-fan And I see it is simple `ws://` instead of `wss://` - what's the story behind that?

But I still don't see why that is preventing the app from running on https (port 443). I myself am the maintainer of [Fidus Writer](https://www.fiduswriter.com) - which runs on https, can be installed as a PWA and itself keeps websocket wss connections to get things from the server.

I really think it would be good to get this app in shape. Is it even open source? If not, maybe it could be made open source so that others can tinker with it? It can be put under a license like the AGPL so that the company is really sure noone else can just run with the code and not share it back.

closed 2020-04-14 by stale[bot]