dresden-elektronik/deconz-rest-plugin #2178: Coordinator unable to join network?

MrJanD, 2019-12-06:
Hi there!

First of all: I tried many Zigbee solutions before (HUE bridge, IKEA Gateway, Zigbee2MQTT) and now I got the ConBee II. Together with FHEM it seems to be working really great for over half a year and I thought I finally got the right solution. THANKS!!
But since a sudden power loss end of last week at night (woke up by a whole shining house with lots of light ;) ) it seems to be broken somehow. Similar problems like [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1596](url) do not really help in my case. 
I tried it with an USB extension cable - noting changed. 
Updated deCONZ docker container - nothing changed.
Updated firmware although I got the latest - noting changed.
Even an empty, fresh docker container with no devices showed the same issue (wierd: Channel and other settings seem to be the same as before, even though everything shall be up like new with no relicts!?).
I was so desperate, I already bought another ConBee II USB gateway, maybe it's broken. 

However, I will try to fix the problem until the new one arrives... 
Attached you can find the log with activated info, APS, ZCL, ZDP and OTAU log flags. Moreover, the configuration with a blackened configuration screenshot. 

Any help would be appreciated! Thanks in advance!

[image]
[deconzOut.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3933932/deconzOut.txt)

MrJanD, 2019-12-09:
As it turn out, it was a missing braid breaker on the power supply cable of the raspberry. 
Since it worked before and all of a sudden unscrutable behavior popped up, I expected it to be something similar. Homematic recommended to use a braid breaker when multiple RF solutions are in vicinity or even attached to the host PC. 
However, again great praise for your Zigbee solution and keep on the good work. :)

robertlandes, 2020-01-02:
@MrJanD is this issue resolved for you? If yes could you please close it?

Thanks

closed 2020-01-03 by MrJanD