dresden-elektronik/deconz-rest-plugin #2180: Innr SP 120 keep losing connection

napcode, 2019-12-06:
I have 4 Innr SP 120 connected to a fantastic Conbee 2 plugged into a Khadas VIM2 running Home Assistant / Hass.io (DECONZ addon). I'm also using several sensors (Xiaomi stuff) and lights (Philips, Ikea). All devices are easily detected in the Phoscon App but only the SP 120 are losing connection after around a day. They work perfectly fine in the beginning (toggling power, reading power sensor data) but after an unknown period of time, they just stop reporting any data or reacting. 
Unfortunately, I don't have more data than this. Are there any logs I should look at or collect? I'd appreciate any help.

SwoopX, 2019-12-06:
Do you maybe have a Hue smatr button in use? Just sumbled over an article [article](https://hueblog.de/2019/11/11/sorgt-der-hue-smart-button-fuer-probleme-mit-innr-produkten/) that people seem to have issues after around one day as you described...

napcode, 2019-12-06:
Interesting. That does sound quite similar. I have a Tradfri Remote and a Xiaomi button hooked up as well. The Xiaomi button was a recent purchase and I had issues even before that. I can try removing the Tradfri remote though. 
I forgot to mention in the first post: the SP 120 works again if I unplug them from the outlet for a second. I don't have to pair them again or something.

ACrazyConcept, 2019-12-07:
I have several SP120s and only very rarely had a lost connection for any of them. Also have several Trådfri remotes and Xiaomi buttons. No Hue buttons though.

SwoopX, 2019-12-07:
Well, I just read is coincidently a couple of minutes before noticing this issue. Other than that, I cannot add anything to the discussion. Got like 10 innr plugs and never had anny issues except for once. But I lso do not have any Hue smart buttons or so.

napcode, 2019-12-08:
Thanks for the hints so far. I removed all devices except 1 SP 120 and it still loses connection after a day. Are there any logs I can use to debug? 
I'll probably try a fresh install on another system. Knowing that the devices work for you guys already helps!

napcode, 2020-01-01:
I've did quite a few experiments with different test environments. Looks like this is somehow related to the hardware or kernel. I guess this issue can be closed as I no longer think this is really a DeCONZ issue.

closed 2020-01-01 by napcode

PaulEins, 2022-06-03:
Hello, I have the same problem.

What can you do?