dresden-elektronik/deconz-rest-plugin #2181: Add support for ecosmart zbt-cctswitch-d0001

crazifyngers, 2019-12-06:
These are inexpensive remotes that come with ecosmart zigbee bulbs.  The bulbs are zigbee 3.0 and work with the default light component.  The remotes though are not recognized.  Attached are the cluster info.  Any chance of getting these added? Thanks!

[image]
[image]
[image]

crazifyngers, 2019-12-07:
here is a link to the actual light/remote with an included picture because apparently their product code does not list any results in google.

https://www.homedepot.com/p/EcoSmart-60-Watt-Equivalent-A19-SMART-LED-Light-Bulb-Tunable-White-Starter-Kit-1-Bulb-A9A19A60WESDZR1/310288335

mbafford, 2019-12-17:
See my PR on zigbee2mqtt for a potential implementation guide: https://github.com/Koenkk/zigbee-herdsman-converters/pull/822

And the Home Assistant PR, which doesn't look to be attempting to sanitize the fairly random inputs from the remote.
https://github.com/dmulcahey/zha-device-handlers/pull/211/files

thedannymullen, 2019-12-20:
I picked one of these up it didn’t pair in the version of deconz I had.  Curious is it supported in any beta releases currently?

thedannymullen, 2020-01-16:
Any updates on whether this feature will be added?  Very curious!  I will probably also order another remote but hate throwing the few of these I have in the trash.  The bulbs integrate great with deconz.

closed 2020-05-26 by stale[bot]

raz0rf0x, 2020-06-07:
I still have this problem with DeconZ. Is this issue going to be addressed?

thedannymullen, 2020-06-07:
Which problem is it you are having?