dresden-elektronik/deconz-rest-plugin #2183: Ikea Tradfri on/off not working

amittel, 2019-12-07:
Hello!
Just stepping into the world of zigbee.

I have problems with the switch. In the phoscon app it's not appearing. But when I switch to the old user interface it appears.
Is there anything I can do about it?
I pressed the reset button 4 times and in the phoscon app nothing happens. I searched under "Switches -> Add new switch -> Others".
 I attached some pictures to show my problem. 

Thanks for any help.

Phoscon app not showing the switch:
[image]

Old app showing the switch:
[image]

My gateway version information:
[image]

Smanar, 2019-12-07:
Hi,
And with "add new switch" then "ikea" instead of "other" ?

amittel, 2019-12-07:
There is no on/off switch. Just a dimmer and a switch with four buttons.
The strange thing for me is, that the button is in some way added but only displayed on the old app and not the new one.

Smanar, 2019-12-07:
It something normal, you can have device in deconz but not in the API, you can have device in deconz+api but not in phoscon. The old app display more device than phoscon.

But this one is on supported device, and seem to be supported > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/853

Perhaps a bad inclusion ? Can you retry to include it ? And you are right, it's in "other" according to previous issue.

amittel, 2019-12-08:
Well, no luck so far. I think I tried everything, but the phoscon app just won't show my switch.

I did it in the old app and the switch works.

I also had a look at http://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
and also no luck getting an api key.

Nevertheless now I can mount my blinds and create some schedule. 
Not perfect, but works...

closed 2020-04-14 by stale[bot]