dresden-elektronik/deconz-rest-plugin #2184: Part of node is not visible in GUI.

AdamWeglarz, 2019-12-07:
Hi,

I have issue that part of node is not visible in GUI. They are available on phoscon and works correctly with external systems (I use domoticz)
Using latest deconz version and latest Conbee FW
[image]
[image]
I am missing for example this device.

[image]

Any hint why?

Adam

Smanar, 2019-12-07:
Can you take a look in the api, to have the MAC adress of this device ?
Something like http://<deconz_ip>/api/<api_key>/sensors

You can just have diffrent name.

AdamWeglarz, 2019-12-07:
In API there are all devices and I expected that as they all work correctly in domoticz.

Smanar, 2019-12-07:
Try to change the name in Phoscon, take a look if another device name haven't changed in deconz.
You can see too the name changement in domoticz logs, with more info, mac adress, previous name, ect ...

AdamWeglarz, 2019-12-07:
On domoticz everything is perfectly perfect. Issue is on GUI of deconz.

Smanar, 2019-12-07:
I m not saying something is wrong on domoticz.
I m saying "you need the mac adress to reconize the device and check if it's realy not in deconz" and I give you some way to have this adress.

closed 2019-12-08 by AdamWeglarz