dresden-elektronik/deconz-rest-plugin #2185: home assistant does not reconnect

k1ng440, 2019-12-07:
Hello,
home assistant does not attempt to reconnect after losing connection (such as restarting the pi)

deconz is running on a dedicated RPI3 and home assistant on VM on freenas,

Kane610, 2019-12-07:
If the Pi gets a new ip address you will need to have ssdp integration enabled so deconz can broadcast it's new address to home assistant

k1ng440, 2019-12-07:
> If the Pi gets a new ip address you will need to have ssdp integration enabled so deconz can broadcast it's new address to home assistant

the PI has static IP address.

Kane610, 2019-12-07:
Then I don't see an issue why deconz can't connect. You would need to share logs and configs

k1ng440, 2019-12-07:
i am not saying it does not connect. it just does not reconnect

```
2019-12-08 03:17:00 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"43","r":"sensors","state":{"lastupdated":"2019-12-07T21:17:00","presence":true},"t":"event","uniqueid":"00:15:8d:00:02:23:86:8d-01-0406"}
2019-12-08 03:17:35 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"38","r":"sensors","state":{"lastupdated":"2019-12-07T21:17:34","presence":true},"t":"event","uniqueid":"00:15:8d:00:01:f4:a8:ed-01-0406"}
2019-12-08 03:17:37 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"43","r":"sensors","state":{"lastupdated":"2019-12-07T21:17:37","presence":true},"t":"event","uniqueid":"00:15:8d:00:02:23:86:8d-01-0406"}
2019-12-08 03:17:43 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"43","r":"sensors","state":{"lastupdated":"2019-12-07T21:17:43","presence":true},"t":"event","uniqueid":"00:15:8d:00:02:23:86:8d-01-0406"}
2019-12-08 03:17:57 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"43","r":"sensors","state":{"lastupdated":"2019-12-07T21:17:57","presence":true},"t":"event","uniqueid":"00:15:8d:00:02:23:86:8d-01-0406"}
2019-12-08 03:18:00 INFO (MainThread) [homeassistant.components.automation] Executing update_battery_status_group_members
2019-12-08 03:18:00 INFO (MainThread) [homeassistant.helpers.script] Script update_battery_status_group_members: Running script
2019-12-08 03:18:00 INFO (MainThread) [homeassistant.helpers.script] Script update_battery_status_group_members: Executing step call service
2019-12-08 03:19:00 INFO (MainThread) [homeassistant.components.automation] Executing update_battery_status_group_members

***** I have restarted the RPI ****

2019-12-08 03:19:29 INFO (SyncWorker_10) [oauth2client.transport] Refreshing due to a 401 (attempt 1/2)
2019-12-08 03:19:29 INFO (SyncWorker_10) [oauth2client.client] Refreshing access_token
2019-12-08 03:19:29 INFO (SyncWorker_4) [oauth2client.transport] Refreshing due to a 401 (attempt 1/2)
2019-12-08 03:19:29 INFO (SyncWorker_4) [oauth2client.client] Refreshing access_token
2019-12-08 03:19:29 INFO (SyncWorker_6) [oauth2client.transport] Refreshing due to a 401 (attempt 1/2)
2019-12-08 03:19:29 INFO (SyncWorker_6) [oauth2client.client] Refreshing access_token
2019-12-08 03:19:31 INFO (MainThread) [homeassistant.helpers.script] Running script
2019-12-08 03:19:31 INFO (MainThread) [homeassistant.helpers.script] Executing step call service
2019-12-08 03:19:31 WARNING (MainThread) [homeassistant.components.mqtt.binary_sensor] No matching payload found for entity: Entryway Light Button1 with state topic: entryway_light/stat/RESULT. Payload: , with value template Template("{{value_json.BUTTON1}}")
2019-12-08 03:19:31 WARNING (MainThread) [homeassistant.components.mqtt.binary_sensor] No matching payload found for entity: Entryway Light Switch3 with state topic: entryway_light/stat/RESULT. Payload: , with value template Template("{{value_json.SWITCH3}}")
2019-12-08 03:20:00 INFO (MainThread) [homeassistant.components.automation] Executing update_battery_status_group_members
2019-12-08 03:20:00 INFO (MainThread) [homeassistant.helpers.script] Script update_battery_status_group_members: Running script
2019-12-08 03:20:00 INFO (MainThread) [homeassistant.helpers.script] Script update_battery_status_group_members: Executing step call service
2019-12-08 03:20:43 ERROR (stream_worker) [homeassistant.components.stream.worker] Error demuxing stream: No dts in packet
2019-12-08 03:21:00 INFO (MainThread) [homeassistant.components.automation] Executing update_battery_status_group_members
2019-12-08 03:21:00 INFO (MainThread) [homeassistant.helpers.script] Script update_battery_status_group_members: Running script
```

Kane610, 2019-12-07:
The integration should try to reconnect indefinitely every 15 seconds

Something like this should be printed "Reconnecting to deCONZ in 15" if you enable debug for pydeconz

k1ng440, 2019-12-07:
> The integration should try to reconnect indefinitely every 15 seconds
> 
> Something like this should be printed "Reconnecting to deCONZ in 15" if you enable debug for pydeconz

sadly its not happening for me.

Kane610, 2019-12-08:
That's the legacy implementation. Normally it's the AIOWsclient being used. Are you saying you have configured the integration to use the legacy websocket?

k1ng440, 2019-12-08:
> That's the legacy implementation. Normally it's the AIOWsclient being used. Are you saying you have configured the integration to use the legacy websocket?

oh. i did not configure to use legacy websocket

philsson, 2019-12-14:
Subscribing here as I have the same issue. Not sure yet as to what scenario causes it, but I think it got triggered last time when upgrading home assistant. I am running hassio with deconz as a plugin in, everything running in docker containers on the same host. Deconz is running fine on its own, it is just the connection between it and HA that is down. Glad to help with logs.

k1ng440, 2019-12-15:
Please check WebSocket automatically reconnects after you restart the deconz.

philsson, 2019-12-15:
Do I need to activate any of the advanced debug output controls to get these logs?
I noticed that deConz seems to receive data from HA, but not the other way around last I got this state. I could turn on my lights from HA but HA never got the feedback that they were on. Also my Zigbee sensors, switches etc did not report back to HA.

devmil, 2020-01-08:
I have the exact same issue. Nothing special configured for the integration, dedicated Raspberry Pi, static IP address.
Home Assistant is running in a docker container on a Synology NAS.

When the Pi restarts then all the values from the DEConZ bridge simply don't update for days.

Please let me know when and how I can provide more insight (logs, ...). I'm running Home Assistant version 0.102.2

philsson, 2020-01-08:
> I have the exact same issue. Nothing special configured for the integration, dedicated Raspberry Pi, static IP address.
> Home Assistant is running in a docker container on a Synology NAS.
>
> When the [...]

I have stopped experiencing these problems the last few weeks. I've seen there are some recent changes in the release notes regarding deConz and connectivity. I am currently running  0.103.5 which seems to work flawlessly so far. Maybe you should try updating Hassio

> 0.103.3 Fix update port and api key on deconz discovery config entry u...
> 0.103.4 Fix deconz SSDP updating Hassio discovery

devmil, 2020-01-08:
Thanks for the hint. I updated HA to 0.103.6 (currently latest release) but still see this behavior in a short test.
Will continue to watch it.

Kane610, 2020-01-09:
If you want to try something, try this:

You need to go into the the hass container and change a file 

```
diff --git a/pydeconz/websocket.py b/pydeconz/websocket.py
index c9190ab..b4d3bef 100644
--- a/pydeconz/websocket.py
+++ b/pydeconz/websocket.py
@@ -59,11 +59,12 @@ class AIOWSClient:
         url = f"http://{self.host}:{self.port}"
 
         try:
-            async with self.session.ws_connect(url) as ws:
+            async with self.session.ws_connect(url, heartbeat=15) as ws:
```

Nalith, 2020-01-09:
Having the exact same problem here. Dedicated PI with static IP.

Oliviakrkk, 2020-03-19:
I have almost the same problem. I tracked it a little.  Problem is repeatable using docker compose and `docker-compose up` command. 

Hass is connecting to the deconz:
```
2020-03-19 11:33:50 INFO (MainThread) [homeassistant.setup] Setting up deconz
2020-03-19 11:33:50 INFO (MainThread) [homeassistant.setup] Setup of domain deconz took 0.0 seconds.
2020-03-19 11:33:50 DEBUG (MainThread) [pydeconz.gateway] Sending "get" "None" to "IP "
```
Then getting response:
```
206 2020-03-19 11:33:51 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: {'config': {'UTC': '2020-03-19T11:33:51',
 207             'apiversion': '1.16.0',
 208             'backup': {'errorcode': 0, 'status': 'idle'},
 209             'bridgeid': '00212EFFFF040982',
 210             'datastoreversion': '60',
 211             'devicename': 'ConBee II',
 212             'dhcp': True,
 213             'factorynew': False,
 214             'fwversion': '0x00000000',
 215             'gateway': 'IP',
 216             'internetservices': {'remoteaccess': 'disconnected'},
 217             'ipaddress': 'IP',
 218             'linkbutton': False,
 219             'localtime': '2020-03-19T11:33:51',
 220             'mac': 'MAC',
 221             'modelid': 'deCONZ',
 222             'name': 'ConbeeIIGW',
 223             'netmask': '255.255.255.0',
```
And then getting list of devices. All devices apart from lights.  
Now lights are marked as unavailable in Hass. 

I checked all sensors and switches DO work. 
```
2020-03-19 11:55:03 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"28","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2020-03-19T11:55:03"},"t":"event","uniqueid":"00:15:8d:00:02:f0:79:e8-01-0012"}
2020-03-19 11:55:03 INFO (MainThread) [homeassistant.components.automation] Executing BathroomRightButton
2020-03-19 11:55:03 INFO (MainThread) [homeassistant.components.automation] BathroomRightButton: Running script
2020-03-19 11:55:03 INFO (MainThread) [homeassistant.components.automation] BathroomRightButton: Executing step call service
2020-03-19 11:55:03 WARNING (MainThread) [homeassistant.helpers.service] Unable to find referenced entities light.bathroomspotlight1, light.bathroomspotlight2, light.bathroomspotlight3, light.bathroomspotlight4

```

So the problem is that deconz is not passing any light through the api. I will try to check DEBUG on deconz now.

Oliviakrkk, 2020-03-19:
So  looking at the timestamps deconz allowed connections to api before setting up lights domain:

Logs from Hass:
```
2020-03-19 12:01:03 DEBUG (MainThread) [pydeconz.gateway] Sending "get" "None" to "IP "
2020-03-19 12:01:04 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: {'config': {'UTC': '2020-03-19T12:01:04',
2020-03-19 12:01:35 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'heatsetpoint': 2000.0}" to "IP /sensors/8/config"
2020-03-19 12:01:35 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/sensors/8/config/heatsetpoint': 2000}}]
2020-03-19 12:01:35 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'heatsetpoint': 2000.0}" to "IP /sensors/61/config"
2020-03-19 12:01:35 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/sensors/61/config/heatsetpoint': 2000}}]
```

Deconz:
```
12:01:04:604 New websocket IP:60376 (state: 3) 
12:01:08:207 COM: /dev/ttyACM0 :  (0x1CF1/0x0030)
12:01:08:207 COM: /dev/ttyS0 :  (0x0000/0x0000)
12:01:08:207 GW firmware update select USB device
12:01:08:383 Announced to internet https://phoscon.de/discover
12:01:13:206 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
12:01:13:215 COM: /dev/ttyACM0 :  (0x1CF1/0x0030)
12:01:13:215 COM: /dev/ttyS0 :  (0x0000/0x0000)
12:01:13:215 COM: /dev/ttyACM0 / serialno: 
12:01:13:215 COM: --dev: /dev/ttyACM0 (ConBee II)
12:01:13:216 auto connect com /dev/ttyACM0
12:01:13:232 Serial com connected
...
12:01:13:405 ~Resource() /lights 0x7ffcc9e96008
12:01:13:405 ~Resource() /lights 0x7ffcc9e96008
12:01:13:405 ~Resource() /lights 0x7ffcc9e96008
12:01:13:405 ~Resource() /lights 0x7ffcc9e96008
```

The first mention of resource lights is 10 second after Hass established connection to deconz. 

And calling `deconz.device_refresh` on Hass helped...

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix