dresden-elektronik/deconz-rest-plugin #2186: Issues that need solving (3)

hellcry37, 2019-12-08:
For me my deconz conbee II stick at this moment has 3 issues:

1. when u remove (reset) a device from zigbee network doesn't make that device to leave the network, that device remains connected (bulbs, sensors, switches etc.)
2. after update successfully firmware on osram bulbs (guessing this happens on any device)  new firmware version is not shown even if i restart device or whatever I do
3. sometimes I loose control over all my bulbs (osram and hue) only the bulbs, every other sensor or switch still works ok. I usually have to power off all bulbs to make them work

Using version 2.5.71

ebaauw, 2019-12-08:
> 1. when u remove (reset) a device from zigbee network doesn't make that device to leave the network, that device remains connected (bulbs, sensors, switches etc.)

That's simply not true.  When you reset a device, it deletes its association to the ZigBee network, the PAN ID, the network key, etc.  After a reset, the device is no longer reachable by and no longer can reach other devices on the network.

ZigBee doesn't maintain connections; it simply sends messages over a radio network.  There's no keep alive or anything to check whether devices are still online.  Think walky-talky rather then cellular network.  You only know when some-one is online when they answer.  When they don't, you don't know whether they heard you, but don't feel like answering, whether they ran out of battery, whether they're out of range, or whether they're unconscious or worse.

Each router in the ZigBee mesh network keeps a table of nearby routers.  The lines in the deCONZ GUI are nothing more than a graphical representation of these neighbour tables; the do **not** represent connections.  These tables are filled by responses to neighbour discovery broadcasts and possibly by monitoring traffic.  In my experience, it can take days for a stale entry to be removed.

The problem with resetting (or powering off) a device, is that it doesn't send a "farewell" message.  Its neighbours will still have an entry in their neighbour table; the REST API plugin will still have the associated resources, the GUI will still display the node (and re-display it after you deleted it, since it's re-discovered from the neighbour tables).  Only when trying to communicate with the device and receiving no answer for some extended time, the REST API plugin marks the device with `reachable` false (and the GUI greys out the node).  It still doesn't know whether the device has been reset, or simply powered off.

Morale: you shouldn't reset (nor actually power off) ZigBee devices; you should make the device leave the network, so it does broadcast a farewell message.  Here's the issue: unlike the Hue bridge, the REST API plugin doesn't tell the device to leave when you delete an associated resource.  Also it doesn't delete the other resources associated with the device.

I'm not sure all ZigBee devices implement the leave/reset request, but I know Hue devices do.  That's why you don't have to reset Hue bulbs after deleting the resource from the Hue bridge.  Note that for this to work, the device needs to be online when deleting the resource.  
For fun: try removing a resource for a light when the light is powered down.  After powering up the light, it will still respond to group commands, even though the Hue bridge no longer knows it.

> 2. after update successfully firmware on osram bulbs (guessing this happens on any device) new firmware version is not shown even if i restart device or whatever I do

The GUI is only updated when it receives new data, either because the devices responds to a _Read Attributes_ command, or because the device issues a _Report Attributes_ command.  Typically, the _Read Attributes_ commands are sent by REST API plugin, but only when pairing the device initially.  I think the issue is that the firmware update is done by the (closed source) STD OTAU plugin, unknown to the REST API plugin.  I'm not sure if and how the REST API plugin could detect a firmware update, so it can redo the pairing automatically.

As a workaround, you can re-read the _Basic_ cluster attributes manually in the GUI.  You might also need to re-read the _Device Descriptor_ and _Simple Descriptor(s)_  (from the leftmost of the two drop-down menus in the node), if the firmware update has changed the device from ZLL to ZigBee 3, or has added/removed clusters.  You need to re-pair the device to get resources for any new clusters, and stale resources won't be removed automatically (see point 1).

> 3. sometimes I loose control over all my bulbs (osram and hue) only the bulbs, every other sensor or switch still works ok. I usually have to power off all bulbs to make them work

This is a bitch.  Unfortunately there's multiple causes why this happens, each with their own remedy.  To figure out what's going on, check the deCONZ log for [errors](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log) and use a ZigBee sniffer.  Some of the common causes:
- The gateway no longer can find a route to the device.  Typically messages (attribute reports) from the device still reach the gateway, and the device still responds to group commands.  A power cycle of the device usually remedies this (the device announcement that the device sends after booting seems to repair the neighbour tables).  Sometimes you might need to open the network/search for devices when doing the power cycle.  
I'm not sure what's causing this (this is too low in the ZIgBee stack for me), but it's related to different manufacturers implementing and/or interpreting the standard differently, resulting in different, incompatible route discovery algorithms.  Notably Trådfri differs from Hue; it took several versions of the RaspBee/ConBee firmware, the deCONZ core programme and the REST API plugin before (most of) those issues where ironed out.  I suspect the issue is more prominent with larger networks (> 20 routers).   
[... cut ...] dn't receive an ACK); my Eurotronic Spirit TRVs don't, and need to be power cycled.
- End devices are out of reach of their parent.  Normally the end device should find a new parent, but Xiaomi devices don't.  You need to re-pair them close to where they'll be used.  Also, don't power off routers.
- ZigBee Green Power devices are out of reach of the gateway.  Unlike the Hue bridge, deCONZ doesn't configure ZGP proxies (lights) to forward the ZGP messages.  I think this is still work in progress.

hellcry37, 2019-12-08:
Hello @ebaauw 

First of all let me thank you for your very well done answer, however i have some mentions:

for point 1. even if i agree with you here on most things I am not a zigbee protocol guru, i know basic things abut it and how it works. for me and all not guru zigbee ppl :) is normal when i delete a device, that device will be gone everywhere and should not be in the network anymore forcefully if necessary. ofc the device would be online when i make it leave the network. for me is a problem to delete a device and see that device is still there 4 weeks later, i have one that is not present in my home from almost 3 months and i can still see it in vnc interface, whatever i do i cant make it gone. I even reset gw, still there. that should be unacceptable for everyone

for point 2, as a normal user, if i can update it with deconz i should be able to see new firmware version in interface, is the normal way to do things, now i understand what are the obstacles but still lest not change basic logical things just because we cant do something yet. I should be able to see the firmware version after update no?

and for point 3, thanks i'll try to debug more

also i just remembered i have one more issue with this. i'll update main post with point 4

ebaauw, 2019-12-08:
> is normal when i delete a device, that device will be gone everywhere and should not be in the network anymore forcefully if necessary.  ofc the device would be online when i make it leave the network.

I agree.

> i have one that is not present in my home from almost 3 months and i can still see it in vnc interface, whatever i do i cant make it gone. I even reset gw, still there.

Resetting the gateway would help only, if the stale device information indeed comes from the gateway.  To be sure it doesn't, you might want to hack the database and remove the device from the `devices` table (search for the mac address and make a note of the `device_id`).  Also delete the linked rows (by `device_id`) from the `device_gui` and `device_descriptors` tables.  Make sure to stop deCONZ and to backup the database before hacking it.

If the node still re-appears after that, one of its former neighbouring routers might still have a stale entry.  When in luck, the GUI might show a line to that router.  I'm not sure a power cycle of the router is supposed to clear the neighbour table; otherwise you might need to factory reset and re-pair it.  I've never needed to do that, though.

> I should be able to see the firmware version after update no?

I agree.  I just don't think it can be fixed in the open source REST API plugin.  dresden elektronik have indicated multiple times that they will not open source the deCONZ core programme; I don't know where they stand on the STD OTAU plugin.  I would love to see some more changes there (sorting the colums, selecting the row automatically when selecting a node).

> also i just remembered i have one more issue with this. i'll update main post with point 4

To the OS, the ConBee II is "just" a USB serial port.  Sounds like the onboard bluetooth and ConBee II are trying to use the same hardware resources for the serial port.

When updating the ConBee firmware (or even when resetting it), it disconnects from and reconnects to the OS, and the OS might assign a different device name.  I don't know the NUC, but I've seen reports that some OSs allow for fixed names for USB devices, based on the device manufacturer and model, instead of "floating" names like ttyACM0, assigned based on the order in which devices connect.

hellcry37, 2019-12-08:
unfortunately I've looked into issue #4 and i cant change where deconz conbee II connects, i believe they have to change it in the firmware. also observed i am not the only one with this issue seem that ppl struggle with this one from some time now

ebaauw, 2019-12-08:
> i believe they have to change it in the firmware

That would have to be the firmware of the NUC USB controller, or the NUC kernel.  The ConBee II doesn't create nor choose the Linux/Unix device name.

hellcry37, 2019-12-08:
I don't know, from where i stand nuc bluetooth works ok without conbee II stick and i've pugged conbee stick in nuc not the other way around. I also tried 2 linux distros and more then one kernel so the way i see it this is a nuc firmware issue (witch i do not believe because i saw cases where ppl have this issue and they dont use a nuc) OR conbee firmware issue witch i do believe since it worked ok until some time ago in the same setup, same linux distro etc

like i said, problems started after a conbee ii firmware update.

easybeat, 2020-01-15:
@ebaauw Hi Erik,
I just had this issue last night and solved it like you today:
`I've seen two or three times over the past couple of years that my network was completely useless. The deCONZ GUI couldn't discover the network topology and devices didn't react to direct commands from wireless switches. I haven't been able to determine the cause. I had to power down all my routers, power cycle the RaspBee, and one by one restore power to my routers. Last time was quite a while ago, not sure which version of deCONZ.`

Did you find a reason for this? My setup was running on same firmware since july and no changes since 10 days.

It is very annoying when this happens as it is a big job to do!

Could this be a deconz problem or do you think it is just related to the architecture of a zigbee mesh network?

Thanks for some information.

Kind regards
Beat

ebaauw, 2020-01-15:
@easybeat as I wrote, I did not find the cause.  Also, I did not have to reconnect (reset and re-pair) my devices, “just” power down the routers.

easybeat, 2020-01-16:
@ebaauw Thank you Erik for the information. Two things to understand.
1. Are all main powered devices automatically routers?
2. If I want to use the deconz app to see my mesh networt map, do I need to install VNC on my RaspBee or can I just connect a screen to it? How do you use it?

Thank you for help as I had not the time so far to dive very deep into deconz and so far only used the phoscon app v1 and v2 to connect devices. But I realise now that there are a few things I need to be able to do in deconz app.

Kind regards
Beat

ebaauw, 2020-01-16:
> 1. Are all main powered devices automatically routers?

Usually, but not always.  E.g. my `lumi.ctrl_neutral` wired switches are mains powered, but end devices.  Best check the GUI: routers are yellow nodes; end devices grey.

> 2. If I want to use the deconz app to see my mesh networt map, do I need to install VNC on my RaspBee or can I just connect a screen to it? How do you use it?

You need to have installed Raspbian incl. the graphical environment to run deCONZ with the GUI enabled.  It will show on a monitor connected to the Raspberry Pi's HDMI port, and/or through screen sharing using VNC.  The VNC server is installed by default on Raspbian (when you install the graphical environement); you only need to activate it using `raspi-config` (5. Interfacing Options | P3. VNC).  You can run this this over SSH, if you enabled that; otherwise you need to connect a monitor to setup the VNC server.

Tip: If you have a Mac, you want to setup the VNC server to use a VNC password instead of a UNIX password.  After that, you can use the macOS built-in Screen Sharing client instead of the VNC client.  You can configure Avahi to announce the VNC server over Bonjour, so your Mac will discover the Pi automatically.  I use the [ScreenSharingMenulet](https://apps.apple.com/us/app/screensharingmenulet/id578078659?mt=12) to connect quickly to my Pi-s (and to my headless macOS server).
```
$ sudo cat > /etc/avahi/services/rfb.service <<+
<?xml version="1.0" standalone='no'?>
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<service-group>
	<name replace-wildcards="yes">%h</name>
	<service>
		<type>_rfb._tcp</type>
		<port>5900</port>
	</service>
</service-group>
+
$ sudo systemctl enable avahi-daemon
$ sudo systemctl start avahi-daemon
```

closed 2020-05-22 by stale[bot]