dresden-elektronik/deconz-rest-plugin #2187: Assign switched to individual lights within a group

clutwo, 2019-12-08:
Hi,

I have the following situation:

I have two lights in my bedroom (so in the bedroom group). There is a Tradfri remote as a main switch connected, which should switch between two scenes. "Main light on and small light off", and "Main light off and small light on". 

I also have a small Tradfri Dimmer ( the new kind) at the bedside, with which I would like to control only the small light. (On, Off, Dim up, dim down).

I cannot seem to find a way to do this within deconz, because (and please correct me if I'm wrong):

 - Switches can only be assigned to groups as a whole
 - Other than call scenes, there is no way to differentiate between lights in groups.
 - Now i need one switch, that controls both lights (via scene for example), and one switch that only 
   dims/toggles one light on/off. 

Is there any way to achieve this from within the deconz webapp?

(These are all IKEA devices, and I'm running the official docker image on a Raspberry Pi 3B+ with a Conbee II)

ebaauw, 2019-12-08:
Each switch should be linked to its own group (check `config.group` of the ZHASwitch resource.  Simply assign only the bedside lamp to the small dimmer's group.

clutwo, 2019-12-08:
Thank you, this worked!! For some reason I thought that adding a light to another group removed it from the first group (I thought I observed that behaviour before, but apparently I'm wrong). 

Another thing, and this might be a warrant a new issue: 

The "Delete Scene" Button in the Scenes editor is almost always set to "display:none;", I only saw it once on my search on how to delete a scene, and thats when I decided to check in the developer console. Is there a reason this button might be set to "display:none"? 

Right now I have to remove "display:none" from the button almost always to delete a scene (or call a click through js console).

danielkihlgren, 2019-12-09:
I can see delete scene when clicking settings in the scene editor.

clutwo, 2019-12-10:
Indeed... Feeling pretty stupid right now.... thanks.

closed 2020-04-16 by stale[bot]