dresden-elektronik/deconz-rest-plugin #2188: deconz phoscon raspbee update not possible

overdrive69, 2019-12-08:
Hi, i tried to update via phoscon webinterface from 26050500 to 26330500.
The update starts, but after the update it shows again the old firmware.

[image]

brusarp, 2019-12-10:
Are you on Hassio with Raspberry?
I've the same error...
Try this:
https://github.com/marthoc/hassio-addons/blob/master/deconz/FIRMWARE-UPGRADE.md
For me worked very good!

salopette, 2019-12-10:
For me also a problem always after reboot

[image]

overdrive69, 2019-12-11:
@brusarp I have raspbian (openhabian). So i think your solution didnt work for me.
@salopette 
For me the problem is different i think. I want to update to a new firmware but after the update it again shows the number of the old firmware.
Does anyone know something

forast3r, 2019-12-18:
Similar issue here, running deconz Hass.io addon, stuck in fw version 26230500.
Have tried several times to upgrade to 26330500 but after update successful, phoscon reconnects to  deconz and the fw version stays in old version.
Any solution?

[image]

overdrive69, 2019-12-18:
So this seams to be a general problem.
Is there a official contact where we can ask for support?

overdrive69, 2019-12-18:
There is an instruction to update the firmware manualy:
/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)

I tried this but i got following error:

/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)

m5x, 2020-01-10:
I get the same error. Looks like `/usr/bin/GCFFlasher_internal.bin` is compiled against version 2.28 of glibc but there is version 2.50 used in the docker container.

m5x, 2020-01-10:
Or `/usr/bin/GCFFlasher_internal.bin` uses `/usr/lib/libwiringPi.so` which requires that version of glibc to be exact.

m5x, 2020-01-10:
I get the same error. Found the fix [here (issue 153)](https://github.com/marthoc/docker-deconz/issues/153#issuecomment-538689683).

mnpg, 2020-03-10:
I get the same error also (RPI3B+, raspbian stretch and Raspbee) -> i've post this fix [here (issue 207)](https://github.com/marthoc/docker-deconz/issues/207#issuecomment-597246324)

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix