dresden-elektronik/deconz-rest-plugin #2190: Osram smart+ plug

alp-by, 2019-12-09:
Hello. I met the problems with adding Osram smart+ plug to my conbee II.
As soon as I connected my plug I was trying to add new device from "switches" but without success. Than I found it out that it was added as light on "lights" area. But it wasn't possible to make it ON or OFF.
Than I removed it from "lights" and trying to add it to "switches" again -> no success. 
The same activities to add it to "lights" -> no success.

I have the latest Phoscon 2.05.71 and conbee II 264A0700.
What is the solution to add device and get it working?
Thank you.

jplopezll, 2019-12-09:
I have some of those and working very well. You must add them as lights, not as switches. Try again as lights and revert.

alp-by, 2019-12-09:
> I have some of those and working very well. You must add them as lights, not as switches. Try again as lights and revert.

Do you connect any "power load" to them during the "adding" light device?

closed 2020-04-16 by stale[bot]

AlbrtoSam, 2020-07-13:
Same issue here!

Smanar, 2020-07-13:
Wich one issue ?

AlbrtoSam, 2020-07-13:
Sorry. my mistake, i realize that i had to add to a group to manage it.