dresden-elektronik/deconz-rest-plugin #2193: Xiaomi wall wired switch with no neutral creating 4 devices when added

xhemp, 2019-12-11:
Bought some Single Button Xiaomi wall wired switches(no neutral) and installed today.
When I add them, I can see that 4 different devices are created:

[image]

Then to change the name, I do it on the last one but the other 3 devices remain there on Phoscon (where the issue probably lies). If I try to delete any of those, all 4 devices are deleted.If I check deCONz, I can see just one device but with multiple clusters:

[image]

The basic cluster is as follows:

[image]

The device works well (no issue there) but the lights part on Phoscon has 9 extra devices ( I have 3 switches).
If you need anything else, please, let me know.

lonebaggie, 2020-02-27:
same issue here. After a while two of bulbs disappear.  The switch works without a problem . If you attempt to delete the phantom bulbs , the switch and all the bulbs are removed.

As per screenshot no manufacturer or switch type is displayed just unknown

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix