dresden-elektronik/deconz-rest-plugin #2194: Update deCONZ software on Windows

smartypantsuk, 2019-12-11:
I noticed the manual update procedure but that seemed to be for the firmware on the Conbee.
I'm using deCONZ on Windows and I see there is a new version of the software, is there any update procedure? How do I know it's been updated, I can't see any versioning information anywhere in the deCONZ application.

Thanks

Smanar, 2019-12-11:
For windows it's easy ^^
Here you can find the last deconz version, just download and install it http://deconz.dresden-elektronik.de/win/
After that go in phoscon , gateway and you will find a button to update firmware if needed.

smartypantsuk, 2019-12-11:
> For windows it's easy ^^
> Here you can find the last deconz version, just download and install it http://deconz.dresden-elektronik.de/win/
> After that go in phoscon , gateway and you will find a button to update firmware if needed.

Thanks, just checking there weren't any pitfalls.
Just updated.

iantait, 2020-02-25:
Semi-hijacking this thread - is 5.73 available for windows as yet (or osx :-)
Thanks.

Smanar, 2020-02-25:
It' s a question ?
Take a look here http://deconz.dresden-elektronik.de/win/

iantait, 2020-02-25:
thanks - it just appeared 🙂

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix