dresden-elektronik/deconz-rest-plugin #2196: Device Support   immax Keyfob-ZB3.0

weltmeyer, 2019-12-11:
[image]
[image]

Currently, the device is not shown in api... dunno, thought i could use it just like a switch :)

weltmeyer, 2020-04-01:
Its working in zigbee2mqtt:
https://www.zigbee2mqtt.io/devices/07046L.html

how to make this working in deconz?

rweise, 2020-04-25:
Is there no way for using this with deconz. I'm using  Bitron Keys but the Immax keyboard is a nice small thing .

weltmeyer, 2020-07-13:
Maybe someone can tell, what is needed to make this working :)

SwoopX, 2020-07-13:
Please open a new device request issue to learn about the requirements (no need to reboo or save), thanks!

emc02, 2020-08-14:
#2774  Device

- Product name: Immax NEO SMART Keyfob
- Manufacturer: Immax
- Model identifier: Keyfob-ZB3.0
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch
  - Other: Keyfob

## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
  - Node Info panel
[image]

  - Basic Cluster attributes in the Cluster Info panel.
[image]

  If available add screenshots of other clusters:
[image]
[image]
[image]
[image]
[image]

I hope this Infos helps and the device will be added soon, because I think it's the only Zigbee keyfob available?!
If there are any further infos needed, please let me know and I will post them here if available!
I would also like to test the implementation if neccesary!

emc02, 2020-09-01:
What can be done here to make this work (faster)?
Would need this for about 100 Installations with RaspBees for commercial usage.

Mimiix, 2020-09-01:
@emc02 I think a dev needs the device itself for that. @SwoopX  might be able to answer this better.

SwoopX, 2020-09-01:
Hm, hard to tell. I'd say "just" bending it into a regular remote should be comparably easy, especially if my PR for the button maps gets its ok. However, could still be a gamble if something unusual should occur, what you cannot see remotely and without sniffing the traffic.

emc02, 2020-09-01:
Well I only need the buttons as button 1 to 4 without any special functions.
Is it possible that you make a testversion that includes the remote and I can check the functionality?

rweise, 2020-09-02:
If it helps, where must i send one to bring it to work? I hope it is here in Germany ;-) and if this is  ok, i add a second one 
https://de.aliexpress.com/item/32728254915.html with the same requirements.

emc02, 2020-09-02:
If it's necessary for the developer to have the device physically, then you can give me your address and I will send one to you, or you buy one and I'll pay it. That would be ok for me.

SwoopX, 2020-09-02:
Here are the steps to download and compile my version:

```
sudo systemctl stop deconz
sudo systemctl stop deconz-gui
git clone --branch immaxhttps://github.com/SwoopX/deconz-rest-plugin.git immax
cd immax
qmake && make -j2
sudo cp /usr/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
/usr/bin/deCONZ --dbg-info=2 > debug
```

When you've re-paired the device, check the generated file `debug` for entries containing:
- no button map for: Keyfob-ZB3.0
However, those are only generated if the button map is not correct. Otherwise, you see something like:
- Button 1002 Arm

emc02, 2020-09-03:
what are the prerequested packages for building this? (I am an raspbian buster)

SwoopX, 2020-09-03:
It will not show in Phoscon, you gotta check the API. Just google for deconz rest api.

However, if pairing as sensor, you should get a green ready message. Push the buttons while pairing.

Also, since you run headless, you need to add the corresponding switch to the last command. Its -plattform=minimal or so...

emc02, 2020-09-03:
OK, got it working with
/usr/bin/deCONZ --dbg-info=2 -platform minimal > debug.txt 

debug file:
https://nextcloud.realsmarthome.at/index.php/s/27GHRm7rXryDXfr

I've tried to pair the device and then pressed all 4 buttons (of course not at the same time)

Hope the file helps, please let me know if I can do something else for you

Mimiix, 2020-09-03:
12:40:08:668 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
12:40:08:683 CTRL. 3.27.212:40:08:871 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
12:40:08:871 dev /dev/ttyAMA0 (/dev/ttyAMA0)
12:40:08:872 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
12:40:09:198 parent process -bash
12:40:09:198 gw run mode: normal
12:40:09:198 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
12:40:09:198 sd-card cid: 035344534333324780ffffffff01361b
12:40:09:200 DB sqlite version 3.27.2
12:40:09:202 DB PRAGMA page_count: 30
12:40:09:202 DB PRAGMA page_size: 4096
12:40:09:202 DB PRAGMA freelist_count: 0
12:40:09:202 DB file size 122880 bytes, free pages 0
12:40:09:202 DB PRAGMA user_version: 6
12:40:09:202 DB cleanup
12:40:09:203 DB create temporary views
12:40:09:203 DB view [0] created
12:40:09:204 DB view [1] created
12:40:09:204 DB view [2] created
12:40:09:205 DB view [3] created
12:40:09:205 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
12:40:09:206 sql exec SELECT key FROM config2
12:40:09:206 sql exec SELECT key,value FROM config2
12:40:09:207 Load config UTC: 2020-08-14T12:56:33 from db.
12:40:09:207 Load config announceinterval: 10 from db.
12:40:09:207 Load config announceurl: http://dresden-light.appspot.com/discover from db.
12:40:09:207 Load config apiversion: 2.05.78 from db.
12:40:09:207 Load config bridgeid: 00212EFFFF05C1C4 from db.
12:40:09:207 Load config datastoreversion: 93 from db.
12:40:09:207 Load config dhcp: true from db.
12:40:09:207 Load config discovery: true from db.
12:40:09:207 Load config factorynew: false from db.
12:40:09:207 Load config fwneedupdate: false from db.
12:40:09:207 Load config fwupdatestate: idle from db.
12:40:09:207 Load config fwversion: 0x26520700 from db.
12:40:09:207 Load config gateway: 127.0.0.1 from db.
12:40:09:207 Load config group0: 65520 from db.
12:40:09:207 Load config groupdelay: 50 from db.
12:40:09:207 Load config gwpassword: $1$8282jdkm$UDU21mBuYoOVZ13nOVFMR1 from db.
12:40:09:207 Load config gwusername: delight from db.
12:40:09:207 Load config homebridge: not-managed from db.
12:40:09:207 Load config homebridge-pin:  from db.
12:40:09:207 Load config homebridgepin:  from db.
12:40:09:207 Load config homebridgeupdate: false from db.
12:40:09:207 Load config homebridgeupdateversion:  from db.
12:40:09:207 Load config homebridgeversion:  from db.
12:40:09:207 Load config ipaddress: 192.168.1.71 from db.
12:40:09:207 Load config linkbutton: false from db.
12:40:09:207 Load config localtime: 2020-08-14T14:56:33 from db.
12:40:09:207 Load config mac: 38:60:77:7c:53:18 from db.
12:40:09:207 Load config modelid: deCONZ from db.
12:40:09:207 Load config name: kastlgreissler from db.
12:40:09:207 Load config netmask: 255.0.0.0 from db.
12:40:09:207 Load config networkopenduration: 60 from db.
12:40:09:207 Load config otauactive: false from db.
12:40:09:207 Load config otaustate: off from db.
12:40:09:207 Load config panid: 0 from db.
12:40:09:207 Load config permitjoin: 0 from db.
12:40:09:207 Load config permitjoinfull: 0 from db.
12:40:09:207 Load config port: 80 from db.
12:40:09:207 Load config portalservices: false from db.
12:40:09:207 Load config proxyaddress: none from db.
12:40:09:207 Load config proxyport: 0 from db.
12:40:09:208 Load config rfconnect: 1 from db.
12:40:09:208 Load config rfconnected: false from db.
12:40:09:208 Load config rgbwdisplay: 1 from db.
12:40:09:208 Load config runmode: systemd/headless from db.
12:40:09:208 Load config starterkitid:  from db.
12:40:09:208 Load config swcommit: b7ac16f55294e5928993b2e19129d416f8e2e1a2 from db.
12:40:09:208 Load config swupdatestate: allreadytoinstall from db.
[... 1620 lines cut ...]
10:42:43:898 CTRL db store gui node 00:0d:6f:ff:fe:40:a9:fe
10:42:43:918 saved node state in 2 ms
10:42:43:934 sync() in 15 ms
10:42:43:935 REST API plugin shutting down
10:42:43:935 DB save zll database items 0x000000C1
10:42:43:935 DB saved in 0 ms
10:42:43:943 sync() in 6 ms
10:42:43:946 ~Resource() /sensors 0xb9a900
10:42:43:946 ~Resource() /sensors 0xb9a9e0
10:42:43:946 ~Resource() /sensors 0xb9aac0
10:42:43:946 ~Resource() /sensors 0xb9aba0
10:42:43:946 ~Resource() /sensors 0xb9ac80
10:42:43:946 ~Resource() /sensors 0xb9ad60
10:42:43:946 ~Resource() /sensors 0xb9ae40
10:42:43:946 ~Resource() /sensors 0xb9af20
10:42:43:946 ~Resource() /sensors 0xb9b000
10:42:43:946 ~Resource() /sensors 0xb9b0e0
10:42:43:946 ~Resource() /sensors 0xb9b1c0
10:42:43:946 ~Resource() /sensors 0xb9b2a0
10:42:43:946 ~Resource() /sensors 0xb9b380
10:42:43:946 ~Resource() /sensors 0xb9b460
10:42:43:946 ~Resource() /lights 0xca3d70
10:42:43:946 ~Resource() /lights 0xca3e28
10:42:43:947 ~Resource() /groups 0xa461b8
10:42:43:947 ~Resource() /config 0xbb8824

SwoopX, 2020-09-03:
Hm, I don't see any binding of the IAS ACE cluster. Can you please try to remove the battery, wait a couple of mins and collect some new debug output while inserting the battery again?

emc02, 2020-09-03:
did you check my debuglog too?
Here once again with 2 keyfobs and both with reinsterted battery and paired a couple of times:
https://nextcloud.realsmarthome.at/index.php/s/Hk8aDEQkQtJQWTF

SwoopX, 2020-09-03:
Basically yes, but the binding must be on cluster 0x0501...

emc02, 2020-09-04:
And what can be done to make this happen?

SwoopX, 2020-09-04:
That is the key question. I can try to add a change later that hopefully forces that

emc02, 2020-09-04:
What I have seen on pairing in Windows GUI to get the screenshots from the first request:
It took long time till the devices appered in the grafic and then not all informations where shown
I can try to reproduce this in windows and maybe I find a solution how to get this done on raspbian.
You said, you need the entry with cluster 0x0501, so I will try that

emc02, 2020-09-04:
tried everything, but can't get the infos from my first post again, don't know why it's that hard to get the deivices on the right cluster?!

SwoopX, 2020-09-04:
That's the reason why remote integration is not exactly fun, especially for more "exotic" devices. I cannot see what's going on in the air here and only work based on assumptions... Anyway, you might want to try this here in addition:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3004#issuecomment-655441969

> tried everything, but can't get the infos from my first post again, don't know why it's that hard to get the deivices on the right cluster?!

What is that supposed to mean? You probably cannot influence anything here.
I'll see if it helps to implement a check for the missing binding after each button press. Maybe that does the trick.

emc02, 2020-09-05:
with adding 0x0501 to in_clusters i've got that:
[image]

how to edit this value on headless raspbian?

SwoopX, 2020-09-05:
The cluster is added to the coordinator, you'd need to expand the cluster list for the blue node by clicking on the rightmost black bullet.

I also noted in your screenshot that the remotes are paired incorrectly. They are missing the 2nd bullet on the right. It's probably best to delete the corresponding sensors via API and then reset and re-pair.

emc02, 2020-09-05:
the screenshot is from my Windows machine, that is using the original Version (not yours)
with another try I got this:
[image]
with this log on windows:
https://nextcloud.realsmarthome.at/index.php/s/Y5EriKy4ffJzXdZ

Another run on raspian headless I got a segmentation fault on pairing the first keyfob:
https://nextcloud.realsmarthome.at/index.php/s/azWy3aCJSDFsHWf

Then it worked and I got this for pairing all 3 keyfobs:
https://nextcloud.realsmarthome.at/index.php/s/8e8mwbF9Sre44fX

SwoopX, 2020-09-06:
Hm, can you please try a manual binding? Take the IAS ACE cluster as source and endpoint 1 from the coordinator as destination.

emc02, 2020-09-07:
How can I do a manual bindibg with debian headless and the rest-api?

SwoopX, 2020-09-07:
You can't, you need the GUI for that. Check out the wiki for possible options to connect.

emc02, 2020-09-08:
ok, then I have to run Raspbian with GUI, need to configurate that, will report back here when finished but maybe someone else can do this in the meantime?

emc02, 2020-09-16:
Hi,
so I've installed the graphical raspbian version now and tried to pair it again with that result:
[2020_09_06_debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5231858/2020_09_06_debug.txt)
[image]

Can you please describe the manual binding process a little bit better. Thanks.

SwoopX, 2020-09-19:
I should receive the keyfob somewhere in the next days so I can have a detailed look.

SwoopX, 2020-09-20:
Binding is usually done as follows, but this is for server clusters, which doesn't apply here as were after a client cluster
[image]
[image]
[image]

emc02, 2020-09-25:
> 
> 
> I should receive the keyfob somewhere in the next days so I can have a detailed look.

perfect! Please tell me when it's ready to test and I'll try it!

emc02, 2020-10-06:
did you sucessfully test the keyfob or is it impossible to use it with deconz?

weltmeyer, 2020-11-14:
still interested :)

Daniel78G, 2020-12-04:
still interested!

hexagoni, 2020-12-14:
Still interested 👍

Mimiix, 2020-12-14:
@SwoopX

SwoopX, 2020-12-14:
Would still require a device for that, I'm afraid :)

hexagoni, 2020-12-14:
> Would still require a device for that, I'm afraid :)

Could send it to you! Or could PayPal you the money to buy it. Please write an email to me to stay in contact. Then we could share our addresses/ contact details.

Mimiix, 2020-12-14:
@hexagoni Best way to reach out is trough Discord! Ping me or swoop there to get in touch ;)

SwoopX, 2020-12-16:
Got the device, should be ready for the next beta.

SwoopX, 2021-02-15:
Added in 2.10.00-beta

closed 2021-02-15 by SwoopX