dresden-elektronik/deconz-rest-plugin #2197: Nordtronic Box Dimmer 2.0 support

scop, 2019-12-11:
I'm wondering if the Nordtronic Box Dimmer 2.0 in-wall switch/dimmer is supported? It's not listed on the supported devices page -- if it's not supported yet, could support for it be added?

https://nordtronic.dk/shop/box-dimmer-2-712p.html

Planning to get some of these, and my ConBee II is on its way, so can't test just yet.

osos, 2019-12-13:
It worked out of the box for me.

addellll, 2019-12-13:
Looks a rebranded Sunricher, and for me Sunricher works great!

jensflorian, 2019-12-13:
https://www.sunricher.com/zigbee-ac-in-wall-switch-sr-zg9101sac-hp-switch.html

I was using one rebranded as "Müller Licht tint" but unable to pair it with phoscon. How do you pair these? The instruction says to turn these on and off five times.

closed 2020-04-18 by stale[bot]