dresden-elektronik/deconz-rest-plugin #2199: More clear error in the logs when the Conbee/Raspbee is not reachable (or wrong serial port is defined).

tubalainen, 2019-12-12:
Would it be possible to print out a more clear message to the default log that the deconz could not communicate with the hardware (serial port) proper? Today its hard to determine that that is the cause.

Example log below (the Conbee has changed port from ttyUSB0 to ttyUSB1).

It does not say that the communication with the hardware failed... Just "device state timeout".

Thanks! :)

```
13:08:15:017 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
13:08:15:114 CTRL. 3.16.213:08:15:272 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
13:08:15:482 parent process /bin/sh
13:08:15:482 gw run mode: docker
13:08:15:482 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
13:08:15:503 DB sqlite version 3.16.2
13:08:15:504 DB PRAGMA page_count: 37
13:08:15:504 DB PRAGMA page_size: 4096
13:08:15:504 DB PRAGMA freelist_count: 2
13:08:15:504 DB file size 151552 bytes, free pages 2
13:08:15:504 DB PRAGMA user_version: 6
13:08:15:504 DB cleanup
13:08:15:505 DB create temporary views
13:08:15:580 don't close database yet, keep open for 900 seconds
13:08:15:582 started websocket server at port 443
13:08:15:587 found node plugin: libde_rest_plugin.so - REST API Plugin
13:08:15:590 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
13:08:15:600 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
13:08:15:798 New websocket 10.11.1.20:59374 (state: 3)
13:08:16:360 device state timeout ignored in state 2
13:08:17:360 device state timeout ignored in state 2
13:08:17:587 New websocket 10.11.1.20:59378 (state: 3)
13:08:18:361 device state timeout ignored in state 2
13:08:19:361 device state timeout ignored in state 2
13:08:20:362 device state timeout ignored in state 2
13:08:21:258 Announced to internet https://phoscon.de/discover
13:08:21:361 device state timeout (handled)
13:08:21:861 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
13:08:21:861 device state timeout ignored in state 1
13:08:22:361 wait reconnect 15 seconds
13:08:23:361 wait reconnect 14 seconds
```

closed 2020-04-18 by stale[bot]

tubalainen, 2020-04-18:
Oh, no response to this issue... Would it be possible to make a more clear error log when a device is not identified?

ebaauw, 2020-04-18:
You need to ask dresden elektronik, the connection to the coordinator device (RaspBee/ConBee I/II) and the corresponding messages are handled by the deCONZ core programme, not by the REST API plugin.