dresden-elektronik/deconz-rest-plugin #2200: Paulmann RGBW 500.47 LED Controller

snador1977, 2019-12-12:
The RGBW controller is recognized but the control does not work correctly. The RGB part can not be controlled baer the W part. If you want in the Phoscon app the W Part control, the white LEDs of the controller flicker briefly and the controller goes to the RGB part. Commonly encountered in the API 2 devices (RGB and W), so it is at least with other RGBW controllers in Deconz. In Paulmann, however, only the RGB part is created.

ebaauw, 2019-12-12:
> Commonly encountered in the API 2 devices (RGB and W), so it is at least with other RGBW controllers in Deconz. In Paulmann, however, only the RGB part is created.

The API plugin creates a resource per ZigBee endpoint, so it’s really up to the light/controller how it advertises itself.  Could you post a screenshot of the endpoints and cluster, see Wiki.

alessandroceleste, 2019-12-14:
I experience the same problem within Phoscon-App. Color temperature can't be adjusted since slider "flies away" when changing to color temperature mode.

[image]

closed 2020-04-19 by stale[bot]