dresden-elektronik/deconz-rest-plugin #2201: Tradfri remote (round / 5 Buttons) - Powerbutton Bug

rehmosch, 2019-12-13:
hey guys,

Servus,

Hello, wanted to ask if someone also has this bug with the Ikea Tradfri power button (round with 5 Buttons): 

If you press the power button briefly you get the value 1002 = all right. 
If you press the power button for a long time you get the value 1001 = all good 
If you hold down the power button for a long time, you do not get the value 1003 = error 
= the error here is, that when you press the "long button" both commands 1002 and 1001 are executed simultaneously (this is the case with all other 4 buttons) not so, here I can execute at least 2 commands separately, but unfortunately not with the power button, does anyone have a solution?

 Thanks Jutta

ebaauw, 2019-12-13:
I agree, an x001 (Hold) event should always be followed by an x003 (Release after hold).  The Trådfri remote sends a command on holding the On/Off button, but not on releasing it.  Imho, the REST API plugin should have ignored the command On hold and not send the 1001.

rehmosch, 2019-12-13:
Dear ebaauw,

thank you for the answer. You think, that the devs can can fix  the REST API bug?  Or i can fix it, when I use the REST API? Sorry, im a noob ;)

ebaauw, 2019-12-13:
You could simply ignore the 1001.  I can create a PR so that the API no longer issues the 1001.  Not sure if any-one is actually using that event, though.

ebaauw, 2019-12-13:
Hm, the command that leads to the 1001 is actually sent on release after hold, not on hold.  So (before my change) the remote sends a command on press, which generates the 1002 and a command on release after hold, which generates the 1001.  It doesn't generate a command on short release (so the press is correctly mapped to 1002), nor on hold.

I'm not sure where I would stand on issuing a 1003 without a prior 1001, but that's far better than issuing a 1001 without a 1003.  However, a single press/hold/release action should not result in a 1002 followed by a 1003, as that would indicate two separate actions.  Still think ignoring the  command on release after hold is the best option.

Note: the Hue dimmer (after which this whole button event scheme is modelled) issues a x000 on press, a x001 every second while holding, a x002 on short release, and a x003 on long release.  So press/release results in x000, x002 and press/hold/release results in x000, x001, ..., x001, x003.  Since API clients that poll the gateway (or the Hue bridge that doesn't support notifications) might miss the x000 or x001 events, it's essential that every sequence ends with a x002 or x003.

Note 2: HomeKit only supports actions on _Single Press_, _Double Press_, and _Long Press_, so homebridge-hue picks up only one buttonevent of each sequence, ignoring the x000 altogether, and ignoring a x001 or x003 after a previous x001 (so the _Short Press_ is generated on the x002 and the _Long Press_ is generated on the first x001 or on the x003).  Unless `hueDimmerRepeat` is set; in that case a _Short Press_ is generated on each x001 and x003, which is great for controlling the volume of my Sonos players through HomeKit.

rehmosch, 2019-12-13:
> 
> 
> Hm, the command that leads to the 1001 is actually sent on release after hold, not on hold ...

Thank you ebaauw,
its not possible that the Power-Button works like the other 4 Buttons? 

My Problem is that the short press  from PowerButton are to controlling my Hue Lights and the long press  is for Sonos to control the Play/Pause Toggle.  Every Time when I hold the Power-Button to Play/Pause my Sonos-Players in same time all my Hue lights going off ;(

I update & upgrade all deconz installations and I will test it later when my daughter is going to sleep.
Thank you

ebaauw, 2019-12-13:
> its not possible that the Power-Button works like the other 4 Buttons? 

You would have to ask IKEA, the remote’s firmware determines when it sends a command.  All the REST API plugin does is mapping those commands to button events.

PremiumUsername, 2019-12-18:
Another bug I've seen is that if you press the lower brightness button once, then hold the power button you will get:

3002     as expected
1002     as expected
4003     very strange and uncalled for, then in quick succession you get 
1001     as expected

> Hm, the command that leads to the 1001 is actually sent on release after hold, not on hold

This is not the case for me, 1001 is sent even if I keep holding in the button (as seen in Home Assistant, I don't know what is going on in lower levels).

smartypantsuk, 2019-12-30:
It's a shame, I purchased the Tradfri remote as an alternative to the hue.
I prefer the Tradfri remote in design but it loses out to the Hue remote in function.
Hope Ikea address these niggles.

closed 2020-05-06 by stale[bot]