dresden-elektronik/deconz-rest-plugin #2202: Hue Motion Sensor return to previous state

traffiqq, 2019-12-13:
Hi, 

after migrating to deconz from my hue bridge i am trying to get my hue motion sensor running.

I already paired it and setup rules in phoscon. Now what i don't know is the following:
In the Hue App i could tell the sensor to return to the previous state instead of switching the lights off, is this possible with deconz too ?

Regards Christian

ebaauw, 2019-12-13:
In theory: yes, Phoscon would need to create similar rules on the deCONZ gateway to what the Hue app creates on the Hue bridge.  These rules probably create a temporary scene to hold the current state on detecting motion, then change the light to the new state.  When no longer detecting motion, they recall the temporary scene.

Phoscon is not open source, and I doubt this will be high on their priority list.  As a workaround, you would need to check the rules the Hue app creates on the Hue bridge, and create similar rules on the deCONZ gateway yourself, using the API.  Unfortunately, the Hue and deCONZ APIs handle scenes differently, so the Hue app cannot be used with deCONZ.

traffiqq, 2019-12-13:
Well, i guess i will try to do that in node red then.

ebaauw, 2019-12-13:
Not sure if node red supports creating a deCONZ scene.  It would need a deCONZ connector/plugin (whatever it's called) instead of a Hue connector to deal with the different API.

traffiqq, 2019-12-13:
No it does not support that but here is my plan:

1. listen for presence in node red
2. save light state in node red
3. trigger the lights
4. start a timer
5. restore old state after timer runs out

I guess that should do the trick.

closed 2020-04-18 by stale[bot]