dresden-elektronik/deconz-rest-plugin #2205: Xiaomi Presence Sensor does not send LUX/ illumination information to conbee 2/ openhab2

cmayr123, 2019-12-13:
Hello,

I hope I post this in the correct way. I am pretty new to smarthome and openhab.

I use the conbee2 stick on a raspberry with openhab 2. I connected the xiaomi presence sensor (RTCGQ01LM) which should send additional information about the illumination status: https://phoscon.de/en/conbee2/compatible

Unfortunately, I can't find a way to get the illumination information :(
Can you help me please?

The REST Api does not show the illumination:

`"7": {
    "config": {
      "battery": 100,
      "duration": 90,
      "on": true,
      "reachable": true,
      "temperature": null
    },
    "ep": 1,
    "etag": "*****",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_motion",
    "name": "Bewegungsmelder",
    "state": {
      "lastupdated": "2019-12-13T14:56:31",
      "presence": false
    },
    "type": "ZHAPresence",
    "uniqueid": "*****"
  }`

Smanar, 2019-12-13:
You have it on another sensor, this one is a ZHAPresence, so it's sonly for presence detection.
You have too a ZHALightLevel somewhere.

cmayr123, 2019-12-13:
Thanks for the response. 
I don't have another sensor like this. Neither via the REST Api, nor in the Openhab Inbox.

fluffymadness, 2019-12-14:
The old mijia sensor doesn't support illumination afaik
https://www.zigbee2mqtt.io/devices/RTCGQ01LM.html
only the aqara model of the sensor has it
https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html

cmayr123, 2019-12-16:
Thanks fluffymadness. This could be the reason! I will get the new model (aqara) and see if this works.

closed 2020-04-23 by stale[bot]