dresden-elektronik/deconz-rest-plugin #2207: inside proxmox lxc container operation not permitted

leschNUX, 2019-12-13:
Hi,

inside a proxmox lxc container with unpriveleged: 0 option, deconz is not starting right. 

deCONZ
-bash: /usr/bin/deCONZ: Operation not permitted

Any idea what is wrong?

SwoopX, 2019-12-14:
Looks like a permission issue. You might want to check if your user is allowed to execute deconz.

jose-d, 2019-12-19:
are you mounting the /dev/acm device inside of the container? Are the permissions right?

you can check my config for inspiration: https://doc.turris.cz/doc/en/public/deconz_lxc_howto

fensoft, 2020-01-17:
I also have this problem on amd64 lxc container. Here is the strace:

```
# strace /usr/bin/deCONZ
execve("/usr/bin/deCONZ", ["/usr/bin/deCONZ"], [/* 19 vars */]) = -1 EPERM (Operation not permitted)
fstat(2, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 5), ...}) = 0
write(2, "strace: exec: Operation not perm"..., 38strace: exec: Operation not permitted
) = 38
getpid()                                = 5520
exit_group(1)                           = ?
+++ exited with 1 +++
```

closed 2020-05-23 by stale[bot]

srd424, 2020-08-17:
In case anyone else stumbles on this .. the postinst in the deconz deb does:
```
setcap cap_net_bind_service,cap_sys_time,cap_sys_boot=+ep /usr/bin/deCONZ
```
In an nspawn container, by default, cap_sys_time and cap_sys_boot are not allowed; deCONZ seems to run fine if you remove those two file capabilities. Possibly same or similar for lxc/lxd.

woody4165, 2021-02-03:
Hi

I'm struggling with installing Conbee II in an LXC contaner.
I followed various guide/forum/messages, but it does not work for me.

I tried these:
https://doc.turris.cz/doc/en/public/deconz_lxc_howto
https://forum.fhem.de/index.php/topic,99210.msg926099.html#msg926099
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2207
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3488

Any suggestion?

Thanks

jose-d, 2021-02-03:
> When I run deCONZ, I get Operation not permitted

there are some default security limits included from ` /usr/share/lxc/config/common.conf` - is this file included in your lxc container config? If yes, try to comment it out.

I mean this line:

```
lxc.include = /usr/share/lxc/config/common.conf
```

woody4165, 2021-02-03:
Nope.
That line is not included in my lxc conf file

```
arch: amd64
cores: 2
hostname: deconz
memory: 2048
net0: name=eth0,bridge=vmbr0,firewall=1,hwaddr=BA:2D:49:AF:D6:2E,ip=dhcp,type=veth
ostype: debian
rootfs: local-lvm:vm-101-disk-0,size=25G
swap: 2048
lxc.cgroup.devices.allow: c 166:* rwm
lxc.mount.entry: /dev/ttyACM0 dev/ttyACM0              none bind,optional,create=file 0 0
lxc.apparmor.profile: unconfined
lxc.cgroup.devices.allow: a
```

srd424, 2021-02-03:
To clarify my earlier comment, to remove `cap_sys_time` and `cap_sys_boot`, you'd need something like:
```
/sbin/setcap cap_net_bind_service+ep /usr/bin/deCONZ
```

woody4165, 2021-02-03:
> To clarify my earlier comment, to remove `cap_sys_time` and `cap_sys_boot`, you'd need something like:
> 
> ```
> /sbin/setcap cap_net_bind_service+ep /usr/bin/deCONZ
> ```

THANKS!

This seems to let it works now!!!
I already did `setcap cap_net_bind_service,cap_sys_time,cap_sys_boot=+ep /usr/bin/deCONZ` and was not working.
Now doing `/sbin/setcap cap_net_bind_service+ep /usr/bin/deCONZ` it runs properly!!!

jose-d, 2021-02-03:
> This seems to let it works now

thx for sharing, I'll update my howto for turris.

woody4165, 2021-02-03:
Now connecting with Phoscon, I get disconnection
After logging in, it appears the page with the image to search a lamp, I click on Proceed without... and then I get this

[image]

and when I go back to the login page I don't find the image with the local conbee 2, that appears after some seconds.
If I try again same procedure, I get same behaviour...

really strange

srd424, 2021-02-03:
If you're using a long-running container that runs unattended-upgrades or similar periodically, be aware that the deconz package postinst may well re-apply those file capabilities on upgrade ..

woody4165, 2021-02-03:
I created yesterday the container with debian 10 template. I just did apt-get update before proceeding .

This is what I get when running manually deCONZ

```
root@deconz:~# deCONZ -platform minimal 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
```

jose-d, 2021-02-03:
> ..

do you see the USB stick in `GCFFlasher_internal -l` (inside of lxc container) ?

jose-d, 2021-02-03:
> deconz package postinst may well re-apply those file capabilities on upgrade ..

true. fortunately, we do have `ExecStartPre` in unit files :)

```
[Unit]
Description=deCONZ: ZigBee gateway -- REST API
Wants=deconz-init.service deconz-update.service
StartLimitIntervalSec=0

[Service]
User=deconz-user
PermissionsStartOnly=true
ExecStartPre=/bin/chown root:dialout /dev/ttyACM0
ExecStartPre=/bin/chmod 664 /dev/ttyACM0
ExecStartPre=/bin/mkdir -p /run/udev/data
ExecStartPre=/sbin/setcap cap_net_bind_service+ep /usr/bin/deCONZ
ExecStartPre=/bin/bash -c "/bin/echo -e 'E:ID_VENDOR_ID=1cf1\nE:ID_MODEL_ID=0030' > /run/udev/data/c166:0"
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80
Restart=on-failure
StartLimitInterval=60
#AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target

```

woody4165, 2021-02-03:
> > ..
> 
> do you see the USB stick in `GCFFlasher_internal -l` (inside of lxc container) ?

Thanks, the stick wasn't seen with GCCFlasher, so I did "Tweak container environment" in the [howto](https://doc.turris.cz/doc/en/public/deconz_lxc_howto)
Strange, because I thoutght I already did...

In any case, now is working really fine!!!!

Last, but not least, now I have to understand how to integrate with Home Assistant. Actually I'm running deCONZ as HA addon, but i would like to have it as a Proxmox Container.

Thanks!!!

woody4165, 2021-02-03:
It seems that if I restart the container GCCFlasher does not see again the Conbee.

I have already edited /etc/systemd/system/

and have it like this:

```
[Unit]
Description=deCONZ: ZigBee gateway -- REST API
Wants=deconz-init.service deconz-update.service
StartLimitIntervalSec=0

[Service]
User=deconz-user
PermissionsStartOnly=true
ExecStartPre=/bin/chown root:dialout /dev/ttyACM0
ExecStartPre=/bin/chmod 664 /dev/ttyACM0
ExecStartPre=/bin/mkdir -p /run/udev/data
ExecStartPre=/sbin/setcap cap_net_bind_service+ep /usr/bin/deCONZ
ExecStartPre=/bin/bash -c "/bin/echo -e 'E:ID_VENDOR_ID=1cf1\nE:ID_MODEL_ID=0030' > /run/udev/data/c166:0"
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80
Restart=on-failure
StartLimitInterval=60
#AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target
```

jose-d, 2021-02-03:
yes, you need to automatize it somehow. ExecStartPre in unit file is one of the possible ways.

woody4165, 2021-02-03:
ok, now it seems that if I restart the CT it works.
But I have to start the service manually, because I get this if I execute service status

```
root@deconz:~# systemctl status deconz
* deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/etc/systemd/system/deconz.service; disabled; vendor preset: enable
   Active: inactive (dead)
lines 1-3/3 (END)
```
Then, if I execute `systemctl start deconz` it works fine without issue.

What should I do to autostart deconz?

Thanks

jose-d, 2021-02-03:
well you have it in "disabled" state - so I'd do 

```
systemctl daemon-reload
systemctl enable deconz
```

woody4165, 2021-02-03:
> well you have it in "disabled" state - so I'd do
> 
> ```
> systemctl daemon-reload
> systemctl enable deconz
> ```

Thanks!!!

Now seems to work ALL fine!!!!