dresden-elektronik/deconz-rest-plugin #2208: Programming Custom Door Lock

ericvandamme, 2019-12-14:
I purchased a Zigbee door lock, and would like to start working on integrating it into deCONZ API. But for the hope of having it merged at some point, is there an expectation to how it should be developed to be accepted? Just curious because I would hate to get locked into a version long term if the door lock could be available for all.

Everything works as expected in deCONZ.

Smanar, 2019-12-14:
If it works fine in deconz its a good starting point.

There is 2 issues with door lock integration, but I think none was finished, if it can answer to your questions
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/579
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/520

ericvandamme, 2019-12-15:
I have read those in the past. But nothing official seemed to come of the conversation. Just guidelines to how it might be approached in the future, including (possibly) new device types.

powermarcel10, 2019-12-15:
Wich zigbee doorlock are you using exactly?

ericvandamme, 2019-12-15:
Schlage Connect Smart Deadbolt with Century Trim in Satin Nickel, Zigbee Certified - BE468GBAK CEN 619 (https://www.amazon.ca/gp/product/B07H3P3JKG/)

closed 2020-04-23 by stale[bot]