dresden-elektronik/deconz-rest-plugin #2209: Xiaomi Two-Way Control Modul Zigbee

MornLum, 2019-12-14:
Hallo,
is the the Xiaomi Two-Way Control Modul Zigbee supported by deCONZ? It is not listed, but nearly all other Xiaomi modules are integrated. 

[https://www.aliexpress.com/item/4000413750000.html?spm=a2g0o.productlist.0.0.7f3515cbl59zUF&s=p&algo_pvid=7f956e4b-3dde-4e37-a154-35b9a08bbd02&algo_expid=7f956e4b-3dde-4e37-a154-35b9a08bbd02-10&btsid=83c79beb-8d78-4988-bf69-59473467b1ef&ws_ab_test=searchweb0_0,searchweb201602_10,searchweb201603_53](url)

If not, is it possible to integrate and how can I support?

alextud, 2019-12-18:
The xiaomi relay is working with deconz. The only downside is missing the power consumption and it's recognised as a dimmable light.

MornLum, 2020-01-03:
Hallo, today, I received two of these switches and tested them. The controller has two channels, however, deconz only finds one channel (channel 2) as a smart plug. In contrast to others, it does not find a second device for the other channel. Also it is not recongnized as a dimmable light. I already deleted it and searched for it again. Again, only the second channel was found. I used the Phoscon App.

Any idea, why it is behaving is this way?

MornLum, 2020-01-04:
Hallo, I have restartet the system and afterwards, both channels have been found.

closed 2020-05-10 by stale[bot]