dresden-elektronik/deconz-rest-plugin #2211: Xiaomi Aqara motion sensor configure delay

Sk4zz, 2019-12-14:
Hi all

I have a handful of the Aqara motion sensors and for some applications they change too slowly from occupied to unoccupied state. It feels like ~20-30 secs  (I haven't stopped the time yet). I just realized that the aqaras have writable `PIR Occupied To Unoccupied Delay` and `Ultrasonic Occupied To Unoccupied Delay` attributes in the Occupancy sensing cluster. The description says:

"The PIROccupiedToUnoccupiedDelay attribute is 16-bits in length and specifies the time delay, in seconds, before the PIR sensor changes **to its occupied** state when the sensed area becomes **unoccupied** ..."

I am a bit confused by the fact that the sensors changes to the occupied state when the area becomes unoccupied. Is this just a mistake or am I missing something?

Also, both are set to 0 but the delay I am experiencing is non zero.

Can someone provide more detailed information about these attributes?

Best regards

ebaauw, 2019-12-14:
The Hue motion sensors also supports the _PIR Occupied to Unoccupied Delay_ attribute.  This is exposed by the REST API plugin as `config.delay`.  Also, the REST API doesn't expose `config.duration` for the Hue motion sensor, as there's no need for deCONZ to delay updating the REST API attribute, because the sensor already implements the delay.

I my experience, it takes ~15 seconds for the Hue motion sensor to recognise "no more motion".  The delay is added to that time, i.e. it delays the setting of _Occupancy_ (the occupied state) to 0, it doesn't change when the sensor hardware detects "no more motion".

You'd have to experiment if the Aqara motion sensor has actually implemented the logic for the delay attribute.  The fact that they also advertise the ultrasonic delay isn't reassuring (unless it actually is an active sensor also using ultrasonic beams to detect motion in addition to passive infrared detection).  If it has, we'd better update the REST API support for the Aqara sensor, exposing `config.delay` and no longer exposing `config.duration`.

Sk4zz, 2019-12-14:
Thank you for the quick reply!

I'll have time tomorrow to experiment and research on the Aqara motion sensors. I somehow doubt that they have an ultrasonic sensor built in, but I'll try to find out.

I already suspected that there is some built in delay to which the attribute(s) just add an additional delay. I was hoping nevertheless that there was a way do reduce this built in delay by updating some attributes in deCONZ...

ebaauw, 2019-12-14:
I don't think there's any attributes to this effect, but there has been mention of a hardware mod to re-detect motion quicker, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1671.

Sk4zz, 2019-12-14:
Nice! Thank you for pointing this out! I might try that over the holidays :smile:

L3H0, 2019-12-16:
> 
> 
> Nice! Thank you for pointing this out! I might try that over the holidays 😄

And this mod works very well.
I highly recommend this.

closed 2019-12-19 by Sk4zz