dresden-elektronik/deconz-rest-plugin #2213: Attribute reporting

Smanar, 2019-12-14:
Hello, I have a problem with attribute reporting on a device, the command works, I have a positive return when using it, but it doesn't work.

So I have used a sniffer to compare the zigbee request for attribute reporting.
Deconz can use only
- min report interval
- max repot interval
- reportable change

And the gateway use
- Timeout period

I can't say if it 's the source of my problem, but which one workaround I can use to make a test  ? Do you think It can be possible only with editing the xml file ?

SwoopX, 2019-12-14:
Not sure if I understand your issue completely here. You do get a attribute reporting response but no attribute changes then?

Smanar, 2019-12-15:
I receive an attribute reporting configuration response but no reporting after.

But I will try today with the cli plugin, using raw command to see if it's the cause of my problem (with mimic the gateway request).

ebaauw, 2019-12-15:
Did you create a binding for the cluster the attribute is in?

Smanar, 2019-12-15:
Yep, I have tried :
- remove all binding.
- remake them
- remake all report attribute using zclattr 0xEEB8 1 0x0B04 06010B055802

06: Configure reporting
050B: attribute
01 : direction
0258 : 600 s

On sniff all is ok, but I have only the pooling, report attribute not working. Perhaps it's because I can't delete the previous attribute reporting that use the other method ?

I have compared wih the original gateway, I realy don't see differences. The attribute is locked to 0, and attribute reporting doesn't work

BTW, it's not a missing feature from deconz, as It don't work too using the second attribute reporting method.

ebaauw, 2019-12-15:
I think you need to specify the 16-bit values in the payload as little-endian.

ma-ca, 2019-12-15:
Perhaps you need to use direction `0x00` to configure sending reporting. The direction `0x01` is used to configure targets to receive reporting.

From the ZCL spec. 2.5.7.1 Configure Reporting Command Frame Format

2.5.7.1.2 Direction Field

Direction Field | Destinations
-- | --
0x00 | The receiver of the Configure Reporting command SHALL Configure Reporting to send to each destination as resolved by the bindings for the cluster hosting the attributes to be reported.
0x01 | This indicates to the receiver of the Configure Reporting command that the sender has configured its reporting mechanism to transmit reports and that, based on the current state of the sender’s bindings, the sender will send reports to the receiver.

Smanar, 2019-12-15:
Same, result.
And I have mimic the gateway, so she is using too exactly the same command.

The last thing I can do is a reset, don't configure reporting using interval, and use only this one with timeout.
There is something that lock the reporting, attribute updating.

Smanar, 2020-01-14:
So I have noticied something on sniffer.
To use attribute reporting with just timeout, the direction is 01
>zclattr 0x7013 1 0x0006 060100005802

To use attribute reporting with min/max/change, the direction is 00
>zclattr 0x7013 1 0x0008 060000002005002C0101

If I try to change it, I have an error, malformed data in request.

BTW I have a problem, I m trying to have a clean device without reporting atribute set, but even with api disabled deconz set standard value 0/600/1.

There is a way to prevent deconz setting itself some value, or a way to erase them ?

ebaauw, 2020-01-14:
The reporting configuration is stored in the device's non-volatile memory.  It's erased on a factory reset of the device, or indeed by writing all Fs.

Smanar, 2020-01-19:
Ok, so I have some reliable information this time.

- It's the device itself that make attribute reporting request with timeout, the zigate answer "unknow command", the conbee doesn't answer at all (need to make more check for conbee, need more sniffer device ...)
- All the gateways, zigate/legrand/conbee make attribute reporting request with min/max/change only.

It's for that the direction is not the same for both request.

But unfortunately for me, it mean it's not the broken request that block my device ....

closed 2020-03-24 by Smanar