dresden-elektronik/deconz-rest-plugin #2214: Changing color of light in scene through API

anjos, 2019-12-14:
I'm trying to use the API to change the light setup of a scene (I have the latest deconz software installed).  I can see that both the brightness and on/off state are stored fine for all lights involved, **but the color of the light is not**.

To change the light setup in a scene, I send (https://dresden-elektronik.github.io/deconz-rest-doc/scenes/#modify):

```
PUT 'http://phoscon:80/api/DEADBEEF/groups/9/scenes/1/lights/26/state' data: {"transitiontime": 0, "on": true, "bri": 51, "xy": [0.326, 0.333]}
```

Which leads to:

```
Response:
[
  {
    "success": {
      "id": "1"
    }
  }
]
```

Since the API returns success, it would imagine the setup worked fine.  Recalling the scene now shows me that the color of the light in question does not change as expected.

Here is the relevant information for the light in question:

```
{
    "26": {
        "ctmax": 500,
        "ctmin": 153,
        "etag": "f767f89dbe3813c7713e7dc8c785cc25",
        "hascolor": true,
        "manufacturername": "Philips",
        "modelid": "LST002",
        "name": "Office Window",
        "state": {
            "alert": "none",
            "bri": 51,
            "colormode": "xy",
            "ct": 369,
            "effect": "none",
            "hue": 8406,
            "on": true,
            "reachable": true,
            "sat": 143,
            "xy": [
                0.4591,
                0.4106
            ]
        },
        "swversion": "1.24.0_r20239",
        "type": "Extended color light",
        "uniqueid": "00:17:88:01:04:4b:8e:a0-0b"
    }
}
```

What is curious is that I can set the state of the light just fine through the API:

```
PUT 'http://phoscon:80/api/DEADBEEF/lights/26/state' data: {"transitiontime": 0, "on": true, "bri": 51, "xy": [0.326, 0.333]}
Response:
[
  {
    "success": {
      "/lights/26/state/on": true
    }
  },
  {
    "success": {
      "/lights/26/state/bri": 51
    }
  },
  {
    "success": {
      "/lights/26/state/xy": [
        0.326,
        0.333
      ]
    }
  }
]
```

This works flawlessly.  What I cannot succeed to do is setting the light color state on a specific scene through the API.  Is that supposed to work?  Any tips would be more than welcome!

Installed version of deconz:

[image]

anjos, 2019-12-20:
Could any guru give some feedback on this, please?

ebaauw, 2019-12-20:
I can give feedback, but not sure you'll like it.  I'm surprised that the API allows updating of light states in a scene.  I don't understand how that translates to ZigBee commands, and would expect that it doesn't work (as per your experience).  Error handling in the API is lousy, so I've learned never trust the return status.

ZigBee scenes are just a number (group ID, scene ID).  When creating/updating/storing a scene, each light stores (the attributes from) its current state (that support scenes) in non-volatile memory under that number.  Upon recalling the scene, the only number is broadcasted and each light restores its state from memory.

The REST API plugin tries keeps a record of each light state for the scene, so that when you recall the scene, it can update its cache immediately, and return the (expected) light states through the API.  This copy is not (hard) linked to the scene states stored in the lights.  Especially when the light is still transitioning while storing the scene, these will be out of sync.  I've seen similar issue on the Hue bridge.

For the update of the light state in a scene to work, the REST API plugin would have to change the light to the new state, wait for the transition to finish, and send the store scene command.  I would have to check the code to be sure, but I'm not sure this is actually done.  Did the light change when you updated the light state in the scene?  Even when it's done, it will be done in the background, after the API call has returned.  You'd definitely need to wait some time (minutes?) in between setting the lights state in the scene and recalling the scene.

I've had several issues trying to create the scenes I wanted, especially with `ct` and `transitiontime`.  I ended up creating a script that does the following:
- Create the scene;
- Set each light to the desired state;
- Sleep, to give the lights enough time to transition to the desired state;
- Store the scene (PUT `/groups/xx/scenes/xx/store`, with a body containing the desired `transitiontime` for restoring the scene).

anjos, 2019-12-20:
Thanks for your feedback - very useful.

I did try as you suggested first ;-), but my program probably did not wait enough time for the transition and also did not work as I would have expected.  Then, I was waiting 10 seconds only.  Maybe too naive...

At the occasion, I also tried to listen to the websocket to let the program realize by itself when it would be OK to call the "store" command.  The problem I faced then was that, if the light does not change state, then the websocket will not report a change after a command.

I wonder if it would be possible to poll for the light state to minimize the wait time.  Any experience with that?

anjos, 2019-12-20:
Let me ask a slightly different question: if one would like to programmatically change all scenes and not use the RESTful API, is there another way to do this?  What would you recommend?

ebaauw, 2019-12-21:
In the deCONZ GUI, open the _Cluster Info_ panel and select a _Scenes_ cluster.  There you can issue the _Store Scene_ command.  You’d have to do this for each light.  If you want to script this, use the CLI plugin, https://github.com/ma-ca/deconz-cli-plugin.  Not for the faint-hearted, though.

> I wonder if it would be possible to poll for the light state to minimize the wait time.

You cannot control when deCONZ (or rather: the REST API plugin) polls a light, but in the GUI, you can manually read the attributes from the _On/Off_, _Level Control_, and _Color Control_ clusters.  The REST API plugin will pick up the light’s responses.

closed 2020-04-26 by stale[bot]