dresden-elektronik/deconz-rest-plugin #2219: No clue how to add Conbee to Harmony Hub

spudje, 2019-12-15:
Hope to find some help here. I'm totally lost. I'm running Phoscon/DeConz (port 8080) and Home assistant Docker images on a Synology NAS. I also have a Hue bridge and a Harmony Hub. I use Hue Bridge only for my lights. I use Conbee for smart plugs, sensors and switches. Everything works greatly together via Home Assistant.

In the Harmony Hub though, I only managed to detect the Hue Bridge. I have no clue how to also add the Conbee?

All help appreciated!

D3nnisd, 2019-12-16:
You cannot.

I had the same issue and i used Home Assistant with a Roku emulation on it. Harmony is able to detect Roku and from there you can configure the roku buttons with your devices on Conbee.

spudje, 2019-12-16:
Oh, OK, so I'm not as stupid as I thought :)

Thanks, I will try the Roku emulation.

D3nnisd, 2020-04-14:
Still a issue!

Mimiix, 2020-06-05:
More discussion here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2486

Will close this one.

closed 2020-06-05 by Mimiix