dresden-elektronik/deconz-rest-plugin #2220: Must all VENDOR_IKEA be forced XY?

Flurkmark, 2019-12-16:
Numerous people have had issues with CT and IKEA lights. There are several bug reports.
This is according to me, caused by a workaround in zcl_tasks.cpp:
```
   useXy = useXy || task.lightNode->manufacturerCode() == VENDOR_IKEA;
```
Seems like the cure is worse than the disease...

technononsense, 2019-12-21:
I'm having issues recalling scenes using ikea CT bulbs and have been looking for a solution (or a workaround). Can you explain what you mean by the above being an issue and whether there's something that you think would work?

Flurkmark, 2019-12-22:
@technononsense 
This issue causes IKEA lights to not report CT in the REST API if you change CT once. Home assistant gets confused and the CT slider does not work. You can still set CT, but not read it. 

If you are using home assistant and scenes there is a bug in IKEA lamps that basically forbids setting CT and brightness at the same time. To workaround use no transition time (ugly) or use scripts instead and set brightness and CT separately with a delay between like this:
```
....
    - service: light.turn_on
      data:
        entity_id: light.light_8
        color_temp: 369
    - service: light.turn_on
      data:
        entity_id: light.light_9
        color_temp: 264
    - delay: '00:00:01'

    - service: light.turn_on
      data:
        entity_id: light.light_10
        brightness: 150
    - service: light.turn_on
...
```
If you are using deconz scenes, I have no idea. The issue is caused by a 'fix' for deconz scenes as far as I can tell..

closed 2020-04-30 by stale[bot]