dresden-elektronik/deconz-rest-plugin #2222: Ghost Device

00lex, 2019-12-16:
Hi! Im using a Raspbee Device and today I see first time a Conbee Instance in my Web Interface, which can't be accessed. No matter if I clear Browser Cache. 

Im on Buster and the actual official Release of deConz. What could it be..?  

[image]

ebaauw, 2019-12-16:
Most likely, your ISP shares “your” public IPv4 address with others, and you’re seeing the gateway of one of your Internet neighbours.

00lex, 2019-12-16:
Why I can see others if Im on my local ip:port of phoscon?

ebaauw, 2019-12-16:
That’s how gateway discovery works.  The Phoscon web app doesn’t know whether it’s being loaded from the gateway or from elsewhere.  It needs to discover the gateway like any other REST API client.  Each gateway registers with the appspot portal using the public IPv4 address.  The web app queries the appspot portal for registered gateways, which returns all registrations under that IPv4 address.

giss69, 2019-12-21:
Hello,

i've suddenly the same issue. 2 more gateways with absolut different Its. Is it possible to block/hide other gateways?
Thank you
[image]

closed 2020-04-26 by stale[bot]